heishamon/HeishaMon #545: Replace HeishaMon small with HeishaMon large (ESP§32)

akuchenbuch, 2024-09-26:
Hi,
I want to replace my small HeishaMon hardware with the ESP32 version. I want to keep all the settings etc.
I am looking for instructions or a white paper, but I can't find anything that will make the exchange as easy as possible for HomeAssistant.
Is there any experience with replacing the hardware?
Both versions have the same firmware version.

Regards
Andreas

geduxas, 2024-09-26:
They are interchangeable without any hassle, if you made some configuration, you should reconfigure new in same way..

dodger421, 2024-10-02:
I’ve done exactly the same thing in the last couple of months.

It’s a simple swap but there isn’t an easy way to transfer settings from the small board to the large board that I could find. You have to manually note any changes you’ve made.

1) Take a note of any settings you’ve changed on the HeishaMon settings webpage (heishamon.local/settings)

2) remove the small board and connect the large board

3) connect to the default local HeishaMon WiFi network

4) browse to http://192.168.4.1/ and configure WiFi (unless using Ethernet on large board), MQTT server and any other settings from your notes

And that’s it. As long as you’ve used the MQTT server and topic base from the small board HomeAssistant will start getting the values automatically. Your entities and automations will all work as they did before.

closed 2024-10-22 by Egyras