heishamon/HeishaMon #546: Heishamon receives no data from Jeisha (tried big and small and different setups)

zeppappa, 2024-09-27:
My 5kW Jeisha is working well, but unfortunately, I can't get the Heishamon to work. Here's my setup:

I ran the 5m cable from Tindie into the house together with the other power and signal cables, and connected the Heishamon (large) inside.
It connects to WiFi successfully, but doesn't receive any data from the Jeisha.
On the first connection attempt, I didn’t turn off the heat pump (could this have caused damage to the Heishamon?). I also suspected the cable might not like being placed next to the power cable, but a second Heishamon (small) I ordered isn't working either.
I even tried connecting it directly with a different cable (short) to the heat pump, but the same issue persists.
Now, I’m stuck and unsure what to do next. Any advice?

The firmware is up to date, I restarted the Jeisha many times and the console shows the following:

```
Thu Sep 26 16:36:23 2024 (291042): Heishamon stats: Uptime: 0 days 0 hours 4 minutes 51 seconds ## Free memory: 78% ## Heap fragmentation: 18% ## Max free block: 24488 bytes ## Free heap: 29928 bytes ## Wifi: 100% (RSSI: -47) ## Mqtt reconnects: 0 ## Correct data: 0.00% Rules active: 0
Thu Sep 26 16:36:23 2024 (291044): Requesting new Panasonic data
Thu Sep 26 16:36:23 2024 (291045): sent bytes: 111 including checksum value: 18
Thu Sep 26 16:36:25 2024 (293047): Previous read data attempt failed due to timeout!
Thu Sep 26 16:36:25 2024 (293047): Received 0 bytes data
```

IgorYbema, 2024-09-27:
Connecting heishamon while the heatpump is running will not harm heishamon. I do this all the time when I need to test new things.
Maybe you have the optional PCB installed? If yes, then you need to connect heishamon to the port on that board and not on the main PCB

zeppappa, 2024-09-28:
Good to know, but I don't have the optional PCB. Only the display,
3-way-valve and water temp sensor are installed. Is there any setting in
the Jeisha?

zeppappa, 2024-09-30:
Does anyone have an idea what I can do to get the heishamon connected?

geduxas, 2024-10-01:
To what pump do you try to connect, and in which connector? Share some photos of your setup.

zeppappa, 2024-10-01:
I have the Panasonic WH-MDC05J3E5 and have bought both the small and big Heishamon and the 5 meter CN-CNT and 1m CN-NMODE cable. On the photo, the big Heishamon is connected via CNT but I have tried all four configurations and also powered the big Heishamon externally. The other photo does not show my HP board, but I wired mine exactly this way. The Heishamon cables I put each into one of the two slots in the blue circle (the one where it fits). Not at the same time, of course. Would it help, if I open it up and take photos of my board?

[image]
[image]

geduxas, 2024-10-02:
Yes i think you should take photos of your board, and check connection, it should be connected to cn-cnt, other connection is for parallel work with cz-taw. Heishamon should get it's power from pump, so no external power source is required. 
Please attach setting screen from heishamon also.

zeppappa, 2024-10-02:
Here is the "standard" setup with the big Heishamon inside the house and 5m cable:
[image]
[image]
[image]

And here is what I tried real quick, to verify that neither the Heishamon is broken, nor the cable is corrupted:
[image]

geduxas, 2024-10-02:
Could it be that your pump hase locked out cn-cnt? Have you tried power cycle your pump (disconnect power by breakers)?

zeppappa, 2024-10-02:
Yes, I always switch it off with breakers when working on it. So many times since I have the problem.

McMagellan, 2024-10-03:
It doesn't look healthy at all how close Heishamon gets to the housing sheet. I hope you used insulation otherwise the circuit board could be damaged.

zeppappa, 2024-10-03:
You are referring to the last picture with the small Heishamon @McMagellan, right? I just plugged it in for testing and took the picture. In my "normal" setup, the big Heishamon is located inside the house in a plastic box. Any idea, why I cannot receive data from the Jeisha, even though I tried multipe setups?

zeppappa, 2024-10-08:
I am sorry to bother, but I would really like to connect the Jeisha to my smarthome. Is there a way to debug the connection to find out where the problem lies? Or should I try other things, like connecting the Panasonic CZ-TAW1B Network Adaptor or installing the CZ-NS4P extension board and connecting Heishamon to that?

jllussa, 2024-10-18:
If you already have a CZ-TAW1B (official Pananasonic Comfort Cloud adapter) you could try it to discard problems on the CNT port of your Aquarea and see if it works ok.

You don't need to buy CZ-NS4P PCB as it won't help you.

Connecting heishamon to Aquarea is really just plug and play (at least to see values from heatpump).

zeppappa, 2024-10-18:
Thanks for the tip @jllussa ! I just ordered CZ-TAW1B and will report back here, if that works. 
I read everywhere that Heishamon is just plug and play and therefore I am so confused. If only there was a way to debug the problem somehow.

McMagellan, 2024-10-20:
I suspect that the port on the motherboard is destroyed.

It would be helpful to carry out a simple test to see whether Heishamon and the cable are working correctly.
Assuming you would create a kabelbridge (or jumper) at the end of the cable between Tx and Rx that is not plugged into the Pana, then Heishamon would receive its own signal again in the loop. 
You would only need to recognize the received data and then switch an output in the console such as: "Pana data loop OK"
@IgorYbema 
@CurlyMoo

zeppappa, 2024-10-26:
So I just installed the CZ-TAW1B Network Adaptor and that works like a charm. I can do everything in Smart- and Service-Cloud. However, I am even more confused now, why Heishamon does not work. I tried both the small and the big with the same cable as the CZ-TAW1B is now using. They must be broken and I'll return them, I guess. Or does anybody have a different idea?

geduxas, 2024-10-26:
How it could be that both are broken in same time? Really it should be hard way to broke them.. 

As you have taw adapter, you could attach one of heishamon parallel in listen mode. Truly i am confused as well.. you should try to reach @IgorYbema about it, maybe he could suggest you to do some testing.

zeppappa, 2024-10-27:
You are right @geduxas - how can they both be broken? But I dont see any other way. I just tried running them in parallel (see screenshot) but Heishamon does not receive any data in listening mode (I tried a bunch of settings again). Im really confused, since noone else seems to have that problem.

[image]

geduxas, 2024-10-27:
No, it's really simple setup, all is required serial connect to pump and power.. in your case something is wrong with serial.. it's not in socket/wire is broken/using wrong wire.. really don't know, it's need to investigate more to hardware.. i think you should reach @IgorYbema if you bought everything from him, maybe he will have more troubleshooting tip's.

zeppappa, 2024-11-04:
I finally solved the riddle. The cn-cnt cable was wrong wired and therefore the 12v destroyed sth on the board. I received a new board, fixed the cable and it works like a charm now. So the

closed 2024-11-04 by zeppappa

IgorYbema, 2024-11-05:
Yes and again sorry for that. You are the second one having this issue. Since the first one I started to check the cables before packaging and haven't seen a bad cable since. So hopefully this will not happen again.

smartlars534, 2026-08-15:
Great—I spent ages trying different things and even bought a new cable. At least now I know what happened, but it still cost me hours of work with no success. :/