heishamon/HeishaMon #548: Heishamon large and CZ-TAW1 at  proxy port

Tom54497, 2024-10-05:
If I use the proxy port for the CZ-TAW1 is there an urgent need to power Heishamon large with additional USB 5V? It works without additional power! 

Any disadvantages?

rvdgaag, 2024-10-05:
Noticed here the same, I'll see what happens. I have an error message now since the upgrade to 3.8 (H43 zone 1 sensor error). Don't know if it is related to the lack of power or a bug. So I downgraded to 3.6 and will observe all weekend.

Tom54497, 2024-10-05:
What did you notice?

The sensor error I got at startup but it vanished after complete installation and rebooting heatpump and Heishamon.

The instructions on tindie.com say:

Powering large heishamon

The large heishamon board can also be powered directly from the heatpump. However if you are planning to use the extra features of this board like ethernet, CZ-TAW1 proxy or the relays, it is best to power the board using an external 5v usb-c power supply as the heatpump can not provide enough current for this to be active all together.

rvdgaag, 2024-10-05:
The Heishamon large and CZ-TAW1 are working in parallel in my setup, both connected to an individual wired ethernet cable.
After downgrading to the latest production version, 3.6, the sensor error didn't come up anymore.
Now I will probably willl connect a 5V usb-c powersupply to the Heishamon next weekend since I'm under the impression the SG contacts on the Panasonic CZ-NS5P optionboard have a lower voltage now, caused by the drain of the Heishamon. Will check with my multi-meter next weekend and when needed install 5V usb-c powersupply.

rvdgaag, 2024-10-05:
**update**, could not wait, the Heishamon works fine powered with the CZ-TAW1 together. The issue with the SG contacts on the CZ-NS5P were caused by me by  faulty wiring from the relay of my smart home system.

I don't use the relays on the Heishamon and the Wifi is not enabled, so not much power is needed obviously.

Tom54497, 2024-10-05:
How do you disable Wifi? There is no toggle switch isn't it!?

rvdgaag, 2024-10-06:
Just don't enter details in de wifi settings and uncheck: 
Enable WiFi hotspot when not connected:

Tom54497, 2024-10-06:
That option is only for the heishamon setup access I guess. If you opt out it you have to do a factory reset at loss of connection

Polosaz, 2024-10-15:
It's probably a silly question, but to work with the CZTAW1 in proxy mode and avoid Wi-Fi... do I have to connect an Ethernet cable to the CZTAW and another to the Heishamon and connect both devices to the router? Or interconnect both devices with an Ethernet cable and connect only one of them to the router?

Polosaz, 2024-10-16:
> Both need their own ethernet cable

Thank you so much. I was commenting on it because I detected an issue, at least in my Aquarea-Heishamon set.

As a summary, in the CZTAW the ethernet port does not work in proxy mode when large Heishamon is powered from a CN-CNMODE cable with an external 5V current source, but it does work when the original CN-CNT cable is used (without an external 5V current source).

More extended:  I had the small Heishamon (in listen only mode) connected using a modified CN-CNMODE cable to have an external 5V power supply with a mobile charger. I had the CZTAW connected via the original CN-CNT cable to the Aquarea and an ethernet cable for the data connection.

I have replaced the small unit with a large Heishamon, initially using that CN-CNMODE cable with an external current supply, a CN-CNT cable to interconnect the Heishamon and the CZTAW and an independent ethernet cable for each of the two units (connected each one to the router). In this configuration the CZTAW does not connect to the router (the second LED light does not turn on), I think because insufficient power for the Ethernet port connection, however the Heishamon connects properly with the router.

If I change this CN-CNMODE cable for the original CN-CNT cable to connect the Aquarea with the Heishamon large (without external power supply) everything works perfectly. This makes sens? it has happened to someone else?

wsowa, 2024-11-11:
If I connect CZ-TAW1 to proxy port of HeishaMon large then can I use heishamon to control the heatpump or does it have to be in listening only mode? It is not clear from the readme... but it reads to me like listening only mode is required only if connected to CN-NMODE port while CZ-TAW1 is on CN-CNT...

closed 2024-12-18 by github-actions[bot]