heishamon/HeishaMon #549: Energy meters

klausj1, 2024-10-06:
I am using the Heishamon together with home assistent.

I have a big difference between the energy data shown directly on the heatpump (heat production / energy consumption per day).

I wanted to follow that up and saw that there is no energy data (kWh) available in the MQTT topics, only the current power. Still, in Home Assistant I have the energy data entities, so probably the Heishamon integration in the Home Assistant calculates the energy (kWh) based on the power measurements (W).

Is this true?

Couldn't the Heishamon provide the energy data on its own? Calcalulate them in the Heishamon would probably be much more accurate than in the home assistent. I assume the daily/weekly/monthly values which are shown in the UI of the heatpump are not available via the protocol? That would be even better.

geduxas, 2024-10-07:
Yes, all calculation's are done outside. Truly pump don't have any power meter (maybe only current consumption of compressor) all power values are predicted (except of heat power) 

And heishamon is only bridge between pump and endpoint, it should do any calculations.

klausj1, 2024-10-07:
At least the Leisha shows power meters in their local UI (german, sorry): 
[image]
These graphics are available per day, per week, and per year. And for production (heat), consumption (electrical enery) and the COP.
So at least it pretends to have production and consumption meters. And it would be fine to have these values in the heishamon.
Maybe the older ones (Jeisha) do not have the meters?

I do not understand what you mean with "all energy values are predicted (except of heat energy)". What do you mean with this, who is predicting? The heatpump? And heat energy is not predicted? Still its not in Heishamon. Sorry, I am lost ...

geduxas, 2024-10-07:
HeatPump don't have power meter's at all. What are you seeing in LCD is calculated locally by controller from same values as we get in heishamon. All power values are accumulated in controller, not in pump itself. So think about this LCD controller as separate device.

As for energy consumption, pump only has some kind of energy usage presets per device, and its really in big offset from true values, so COP and power consumption is not real.

Egyras, 2024-10-08:
@geduxas for newer K and L series as remember its possible to connect 3rd party mater for electrical energy :)

klausj1, 2024-10-10:
Sure, but I have a totally external meter already - a shelly in my contro cabinet

closed 2024-10-10 by klausj1