heishamon/HeishaMon #550: Heishamon does not reconnect automatically after WLAN-Outage

klausj1, 2024-10-06:
I have the Heishamon Firmware 3.8, the small version, connected via WLAN.

Today I had a 5-10 minutes outage of my WLAN, and the Heishamon did not reconnect when it was back again. Instead, it created an own access point to connect to. I connected to this access point with my mobile, and rebooted the Heishamon via UI. Afterwards it was connected again to the WLAN.

Would be good if it reconnects to the WLAN automatically.

jllussa, 2024-10-09:
You have to uncheck "Enable WiFi hotspot when not connected" and Heishamon will reconnect automatically.

If you ever change your wifi then you will have to factory reset the board.
Double click on reset button.
Then hotspot will reappear.

klausj1, 2024-10-10:
Thanks, that was somehow clear. Still it would be great if the Heishamon checks every 5 minutes or so if the configured WLAN is  available again (as long as no client is connected to the hotspot), and if not, open the hotspot again. Then there is no risk that you have to search for the reset button and you do not know how to reset.

IgorYbema, 2024-10-10:
I should auto-reconnect already but in some situations I also noticed that it doesn't. I still don't know why. So for now, just reset the board if this happens.

klausj1, 2024-10-10:
Then I'll leave the "Enable WiFi hotspot when not connected" checked and hope I do not have too many WLAN-problems. Esp. when I am not at home ...

closed 2024-10-10 by klausj1