heishamon/HeishaMon #553: Getting values individually via HTTP

lkakas, 2024-10-12:
Is it possible to get values from heishamon with a http get method one by one or individually instead of a big chunk of json? Like the "command" url which can be used to set params. The problem with the JSON is that there arent uniq rows as each value is in a " value:" row which makes impossible to distinguish between the params with loxone. 

I need this to be able to integrate with loxone.

Thanks

geduxas, 2024-10-12:
json is in array of objects.. so each object has it's values.. i am unfamiliar with loxone, how about mqtt?

lkakas, 2024-10-12:
I can process JSON which contains uniqe names in the structure for the objects: 

[image]

But not like the heishamon:
[image]

Loxone isn't smart enough to interpret the whole json structure :-( 

On the other hand I don't want to install another appliance with MQTT capability.

lkakas, 2024-10-12:
I managed to extract the values from the objects in Json. 

For example with the following "Command recognition"  in Loxone:  

\i"Heatpump_State","Value":\i\v
\i"Operations_Counter","Value":\i\v

closed 2024-10-12 by lkakas