heishamon/HeishaMon #554: Set1: SetHeatpump not working with Panasonic wh-mdc07j3e5

agenstarburst, 2024-10-14:
If I send a mqtt command set1 SetHeatpump=1 on my wh-mdc07j3e5 the command will not be followed. This happens no matter if I use the mqtt sendto block in blockly (iobroker) nor if I send the command via URL command http://192.168.xxx.xx/command?setHeatPump=0. 

There is no fundamental malfunction as other commands (e.g. as Set29 SetHeatingOffOutdoorTemp=15°C) are properly working. Let me know if you need any further data for analysis.

geduxas, 2024-10-15:
So you're trying to set it to 0 or 1? 1 will turn it on from suspension, 0 will bring it off. Also if i remember correctly beep will also be hered in control panel

agenstarburst, 2024-10-15:
I tried both already. So if the pump was off, I tried to switch it on via SetHeatpump=1 and also if it was on, I tried to switch it off by SetHeatpump=0. If I use the URL interface with other commands as e.g. the Set29, I get an acknowledge that the value was set. If I use the SetHeatpump command, the page just stays blank.

geduxas, 2024-10-15:
For me in latest version everything work's fine, so you have issue.. do you run rule's? Try disable rules and then set commands, or maybe you have some retained set commands in mqtt? Try disable mqtt..also send logs from console while executing commant

agenstarburst, 2024-10-15:
It's a fresh setup done in 09/24 using HeishaMon.ino.d1-v3.8.bin, no rules used yet. This command did not work from the first day on. I exported the logfile that was created during sending the command via http://192.168.xxx.xx/command?setHeatPump=0. I sent the command SetHeatingOffOutdoorTemp=16°C before setHeatPump=0 and sent SetHeatingOffOutdoorTemp=15°C after the setHeatPump=0 command so it is easier to identify the timing. 

`Tue Oct 15 09:50:48 2024 (15002): Starting initial wifi scan ...
Tue Oct 15 09:50:48 2024 (15006): Connecting to mqtt server ...
Tue Oct 15 09:50:48 2024 (15081): Heishamon stats: Uptime: 0 days 0 hours 0 minutes 15 seconds ## Free memory: 70% ## Heap fragmentation: 7% ## Max free block: 25336 bytes ## Free heap: 27088 bytes ## Wifi: 50% (RSSI: -75) ## Mqtt reconnects: 1 ## Correct data: 100.00% Rules active: 0
Tue Oct 15 09:50:48 2024 (15099): Requesting new panasonic data
Tue Oct 15 09:50:48 2024 (15106): sent bytes: 111 including checksum value: 18 
Tue Oct 15 09:50:49 2024 (15519): Received 203 bytes data
Tue Oct 15 09:50:49 2024 (15520): Checksum and header received ok!
Tue Oct 15 09:50:49 2024 (15545): received TOP0 Heatpump_State: 1
Tue Oct 15 09:50:49 2024 (15547): received TOP1 Pump_Flow: 0.13
Tue Oct 15 09:50:49 2024 (15548): received TOP2 Force_DHW_State: 0
Tue Oct 15 09:50:49 2024 (15553): received TOP3 Quiet_Mode_Schedule: 0
Tue Oct 15 09:50:49 2024 (15560): received TOP4 Operating_Mode_State: 0
Tue Oct 15 09:50:49 2024 (15567): received TOP5 Main_Inlet_Temp: 25.75
Tue Oct 15 09:50:49 2024 (15572): received TOP6 Main_Outlet_Temp: 26
Tue Oct 15 09:50:49 2024 (15582): received TOP7 Main_Target_Temp: 30
Tue Oct 15 09:50:49 2024 (15584): received TOP8 Compressor_Freq: 0
Tue Oct 15 09:50:49 2024 (15590): received TOP9 DHW_Target_Temp: 52
Tue Oct 15 09:50:49 2024 (15597): received TOP10 DHW_Temp: -128
Tue Oct 15 09:50:49 2024 (15601): received TOP11 Operations_Hours: 245
Tue Oct 15 09:50:49 2024 (15645): received TOP12 Operations_Counter: 216
Tue Oct 15 09:50:49 2024 (15647): received TOP13 Main_Schedule_State: 0
Tue Oct 15 09:50:49 2024 (15648): received TOP14 Outside_Temp: 12
Tue Oct 15 09:50:49 2024 (15656): received TOP15 Heat_Power_Production: 0
Tue Oct 15 09:50:49 2024 (15662): received TOP16 Heat_Power_Consumption: 0
Tue Oct 15 09:50:49 2024 (15668): received TOP17 Powerful_Mode_Time: 0
Tue Oct 15 09:50:49 2024 (15674): received TOP18 Quiet_Mode_Level: 0
Tue Oct 15 09:50:49 2024 (15682): received TOP19 Holiday_Mode_State: 0
Tue Oct 15 09:50:49 2024 (15686): received TOP20 ThreeWay_Valve_State: 0
Tue Oct 15 09:50:49 2024 (15692): received TOP21 Outside_Pipe_Temp: 13
Tue Oct 15 09:50:49 2024 (15699): received TOP22 DHW_Heat_Delta: -8
Tue Oct 15 09:50:49 2024 (15744): received TOP23 Heat_Delta: 2
Tue Oct 15 09:50:49 2024 (15746): received TOP24 Cool_Delta: 5
Tue Oct 15 09:50:49 2024 (15747): received TOP25 DHW_Holiday_Shift_Temp: -15
Tue Oct 15 09:50:49 2024 (15753): received TOP26 Defrosting_State: 0
Tue Oct 15 09:50:49 2024 (15760): received TOP27 Z1_Heat_Request_Temp: -1
Tue Oct 15 09:50:49 2024 (15767): received TOP28 Z1_Cool_Request_Temp: 0
Tue Oct 15 09:50:49 2024 (15772): received TOP29 Z1_Heat_Curve_Target_High_Temp: 40
Tue Oct 15 09:50:49 2024 (15782): received TOP30 Z1_Heat_Curve_Target_Low_Temp: 31
Tue Oct 15 09:50:49 2024 (15786): received TOP31 Z1_Heat_Curve_Outside_High_Temp: 5
Tue Oct 15 09:50:49 2024 (15794): received TOP32 Z1_Heat_Curve_Outside_Low_Temp: -12
Tue Oct 15 09:50:49 2024 (15802): received TOP33 Room_Thermostat_Temp: 23
Tue Oct 15 09:50:49 2024 (15846): received TOP34 Z2_Heat_Request_Temp: 0
Tue Oct 15 09:50:49 2024 (15847): received TOP35 Z2_Cool_Request_Temp: 0
Tue Oct 15 09:50:49 2024 (15849): received TOP36 Z1_Water_Temp: -78
Tue Oct 15 09:50:49 2024 (15855): received TOP37 Z2_Water_Temp: -78
Tue Oct 15 09:50:49 2024 (15862): received TOP38 Cool_Power_Production: 0
Tue Oct 15 09:50:49 2024 (15868): received TOP39 Cool_Power_Consumption: 0
Tue Oct 15 09:50:49 2024 (15874): received TOP40 DHW_Power_Production: 0
Tue Oct 15 09:50:49 2024 (15882): received TOP41 DHW_Power_Consumption: 0
Tue Oct 15 09:50:49 2024 (15887): received TOP42 Z1_Water_Target_Temp: 30
Tue Oct 15 09:50:49 2024 (15894): received TOP43 Z2_Water_Target_Temp: 38
Tue Oct 15 09:50:49 2024 (15901): received TOP44 Error: No error
Tue Oct 15 09:50:49 2024 (15947): received TOP45 Room_Holiday_Shift_Temp: -15
Tue Oct 15 09:50:49 2024 (15948): received TOP46 Buffer_Temp: -78
Tue Oct 15 09:50:49 2024 (15950): received TOP47 Solar_Temp: -78
Tue Oct 15 09:50:49 2024 (15959): received TOP48 Pool_Temp: -78
Tue Oct 15 09:50:49 2024 (15961): received TOP49 Main_Hex_Outlet_Temp: 25
Tue Oct 15 09:50:49 2024 (15967): received TOP50 Discharge_Temp: 33
Tue Oct 15 09:50:49 2024 (15974): received TOP51 Inside_Pipe_Temp: 18
Tue Oct 15 09:50:49 2024 (15982): received TOP52 Defrost_Temp: -31
Tue Oct 15 09:50:49 2024 (16011): received TOP53 Eva_Outlet_Temp: 17
Tue Oct 15 09:50:49 2024 (16012): received TOP54 Bypass_Outlet_Temp: -31
Tue Oct 15 09:50:49 2024 (16041): received TOP55 Ipm_Temp: 19
Tue Oct 15 09:50:49 2024 (16043): received TOP56 Z1_Temp: 26
Tue Oct 15 09:50:49 2024 (16044): received TOP57 Z2_Temp: -128
Tue Oct 15 09:50:49 2024 (16048): received TOP58 DHW_Heater_State: 0
Tue Oct 15 09:50:49 2024 (16054): received TOP59 Room_Heater_State: 0
Tue Oct 15 09:50:49 2024 (16061): received TOP60 Internal_Heater_State: 0
[... 113 lines cut ...]
[... cut ...]
Tue Oct 15 09:51:23 2024 (50014): Heishamon stats: Uptime: 0 days 0 hours 0 minutes 50 seconds ## Free memory: 75% ## Heap fragmentation: 12% ## Max free block: 25336 bytes ## Free heap: 28736 bytes ## Wifi: 48% (RSSI: -76) ## Mqtt reconnects: 1 ## Correct data: 84.62% Rules active: 0
Tue Oct 15 09:51:23 2024 (50032): Requesting new panasonic data
Tue Oct 15 09:51:23 2024 (50034): sent bytes: 111 including checksum value: 18 
Tue Oct 15 09:51:23 2024 (50430): Received 203 bytes data
Tue Oct 15 09:51:23 2024 (50432): Checksum and header received ok!
Tue Oct 15 09:51:23 2024 (50434): received TOP5 Main_Inlet_Temp: 25.50
`

geduxas, 2024-10-15:
I don't see any commands between 16 and 15 values, wait around 5s between commands, also turn on debug log

agenstarburst, 2024-10-15:
I don't see the command either. Suppose that is the reason why it's not working. Question is, why is the command noch published via heishamon? 

"Debug log to MQTT topic from start" was activated in settings. 

Tue Oct 15 17:30:59 2024 (27625355): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 25 seconds ## Free memory: 73% ## Heap fragmentation: 10% ## Max free block: 25336 bytes ## Free heap: 28112 bytes ## Wifi: 48% (RSSI: -76) ## Mqtt reconnects: 6 ## Correct data: 99.64% Rules active: 0
Tue Oct 15 17:30:59 2024 (27625374): Requesting new panasonic data
Tue Oct 15 17:30:59 2024 (27625376): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:30:59 2024 (27625806): Received 203 bytes data
Tue Oct 15 17:30:59 2024 (27625808): Checksum and header received ok!
Tue Oct 15 17:30:59 2024 (27625811): received TOP49 Main_Hex_Outlet_Temp: 28
Tue Oct 15 17:31:04 2024 (27630356): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 30 seconds ## Free memory: 70% ## Heap fragmentation: 7% ## Max free block: 25336 bytes ## Free heap: 27088 bytes ## Wifi: 50% (RSSI: -75) ## Mqtt reconnects: 6 ## Correct data: 99.64% Rules active: 0
Tue Oct 15 17:31:07 2024 (27633325): Requesting new panasonic data
Tue Oct 15 17:31:07 2024 (27633325): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:07 2024 (27633339): set heating off outdoor temp 6
Tue Oct 15 17:31:07 2024 (27633341): Already sending data. Buffering this send request
Tue Oct 15 17:31:07 2024 (27633802): Received 203 bytes data
Tue Oct 15 17:31:07 2024 (27633803): Checksum and header received ok!
Tue Oct 15 17:31:07 2024 (27633805): received TOP49 Main_Hex_Outlet_Temp: 27
Tue Oct 15 17:31:07 2024 (27633813): Sending command from buffer
Tue Oct 15 17:31:07 2024 (27633816): sent bytes: 111 including checksum value: 12 
Tue Oct 15 17:31:08 2024 (27634290): Received 203 bytes data
Tue Oct 15 17:31:08 2024 (27634291): Checksum and header received ok!
Tue Oct 15 17:31:09 2024 (27635356): Lost MQTT connection!
Tue Oct 15 17:31:09 2024 (27635357): Reconnecting to mqtt server ...
Tue Oct 15 17:31:09 2024 (27635371): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 35 seconds ## Free memory: 62% ## Heap fragmentation: 8% ## Max free block: 22096 bytes ## Free heap: 23888 bytes ## Wifi: 52% (RSSI: -74) ## Mqtt reconnects: 7 ## Correct data: 99.64% Rules active: 0
Tue Oct 15 17:31:09 2024 (27635390): Requesting new panasonic data
Tue Oct 15 17:31:09 2024 (27635393): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:09 2024 (27635850): Received 203 bytes data
Tue Oct 15 17:31:09 2024 (27635852): Checksum and header received ok!
Tue Oct 15 17:31:09 2024 (27635856): received TOP77 Heating_Off_Outdoor_Temp: 6
Tue Oct 15 17:31:09 2024 (27635925): sending raw value
Tue Oct 15 17:31:09 2024 (27635927): sent bytes: 1 including checksum value: 0 
Tue Oct 15 17:31:11 2024 (27637694): sending raw value
Tue Oct 15 17:31:11 2024 (27637696): Already sending data. Buffering this send request
Tue Oct 15 17:31:11 2024 (27637930): Previous read data attempt failed due to timeout!
Tue Oct 15 17:31:11 2024 (27637932): Received 0 bytes data
Tue Oct 15 17:31:11 2024 (27637933): Sending command from buffer
Tue Oct 15 17:31:11 2024 (27637938): sent bytes: 1 including checksum value: 0 
Tue Oct 15 17:31:13 2024 (27639947): Previous read data attempt failed due to timeout!
Tue Oct 15 17:31:13 2024 (27639949): Received 0 bytes data
Tue Oct 15 17:31:14 2024 (27640358): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 40 seconds ## Free memory: 73% ## Heap fragmentation: 10% ## Max free block: 25336 bytes ## Free heap: 28168 bytes ## Wifi: 52% (RSSI: -74) ## Mqtt reconnects: 7 ## Correct data: 99.60% Rules active: 0
Tue Oct 15 17:31:14 2024 (27640376): Requesting new panasonic data
Tue Oct 15 17:31:14 2024 (27640379): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:14 2024 (27640824): Received 203 bytes data
Tue Oct 15 17:31:14 2024 (27640826): Checksum and header received ok!
Tue Oct 15 17:31:14 2024 (27640829): received TOP49 Main_Hex_Outlet_Temp: 28
Tue Oct 15 17:31:19 2024 (27645359): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 45 seconds ## Free memory: 73% ## Heap fragmentation: 10% ## Max free block: 25336 bytes ## Free heap: 28120 bytes ## Wifi: 50% (RSSI: -75) ## Mqtt reconnects: 7 ## Correct data: 99.60% Rules active: 0
Tue Oct 15 17:31:19 2024 (27645377): Requesting new panasonic data
Tue Oct 15 17:31:19 2024 (27645380): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:19 2024 (27645797): Received 203 bytes data
Tue Oct 15 17:31:19 2024 (27645799): Checksum and header received ok!
Tue Oct 15 17:31:24 2024 (27650360): Heishamon stats: Uptime: 0 days 7 hours 40 minutes 50 seconds ## Free memory: 73% ## Heap fragmentation: 10% ## Max free block: 25336 bytes ## Free heap: 28088 bytes ## Wifi: 40% (RSSI: -80) ## Mqtt reconnects: 7 ## Correct data: 99.60% Rules active: 0
Tue Oct 15 17:31:24 2024 (27650379): Requesting new panasonic data
Tue Oct 15 17:31:24 2024 (27650381): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:24 2024 (27650771): Received 203 bytes data
Tue Oct 15 17:31:24 2024 (27650773): Checksum and header received ok!
Tue Oct 15 17:31:24 2024 (27650776): received TOP49 Main_Hex_Outlet_Temp: 27
Tue Oct 15 17:31:24 2024 (27650779): received TOP50 Discharge_Temp: 25
Tue Oct 15 17:31:24 2024 (27650786): received TOP51 Inside_Pipe_Temp: 22
[... cut ...] i: 42% (RSSI: -79) ## Mqtt reconnects: 7 ## Correct data: 99.60% Rules active: 0
Tue Oct 15 17:31:34 2024 (27660380): Requesting new panasonic data
Tue Oct 15 17:31:34 2024 (27660383): sent bytes: 111 including checksum value: 18 
Tue Oct 15 17:31:34 2024 (27660815): Received 203 bytes data
Tue Oct 15 17:31:34 2024 (27660817): Checksum and header received ok!
Tue Oct 15 17:31:34 2024 (27660819): received TOP6 Main_Outlet_Temp: 27.50
Tue Oct 15 17:31:34 2024 (27660829): received TOP56 Z1_Temp: 27

agenstarburst, 2024-10-15:
I haven't tried all other MQTT command topics, but 5-6 that I have tried did work as expected. Only the setheatpump command does not work. 

Want to use it for deactivating the heat pump when I receive high pricing from Tibber.

geduxas, 2024-10-15:
Oh, i missed fact that you trying **set** command instead of **Set**! Commands are case sensitive. So that's why not working for you

agenstarburst, 2024-10-15:
Unbelievable. I must have typed it incorrectly the first time I tried and from then on the browser always took the wrong command from history. Thank you so much!!!

geduxas, 2024-10-15:
Yes, because in heatpump telemetry data all values are +1 

So if issue is resolved, please don't forget to close

closed 2024-10-15 by agenstarburst