heishamon/HeishaMon #555: Execute rules even after a crash reboot

stumbaumr, 2024-10-16:
Hi,
as I still get crash reboots sometimes which disables my otherwise correctly working rules I propose an option in the settings:
"Disable rules on crash reboot" with default on.

My uptime is high, but I still have to look like every day at least once if the rules are still active.

[image]

stumbaumr, 2024-11-02:
@IgorYbema , @CurlyMoo , any thoughts on this?

CurlyMoo, 2024-11-02:
Good idea. But i want to suggest to having it by default off.

stumbaumr, 2024-11-02:
So the text would be "Enable rules on crash reboot" with default off?

stumbaumr, 2024-11-02:
Most users with rules have no crash problem - so they do not need that setting.
On the other hand if they experience a crash boot loop caused by their rules, they want their rules disabled on crash by default. My and McMagellans rules can crash - but the cause is known and accepted. If we get a crash boot loop we know how to reset the device to unset that setting.

So my belief is that it should be by default off (no rules activated after crash reboot).

CurlyMoo, 2024-11-03:
PR is made:
https://github.com/IgorYbema/HeishaMon/pull/143

stumbaumr, 2024-12-04:
This should be merged…

erikrozendaal, 2024-12-18:
Please merge this. I've a set of rules that runs stably for multiple days but then crashes somewhere in the rules (no idea why), so it should be safe to reload the rules in this case.

stumbaumr, 2025-01-18:
Has been merged and will arrive with version 3.9.

closed 2025-01-18 by stumbaumr