heishamon/HeishaMon #556: Opentherm Read values outsideTemp, inletTemp, outletTemp and dhwTemp are integers instead of floating point

blb4github, 2024-10-16:
According readme the Opentherm Read values outsideTemp, inletTemp, outletTemp and dhwTemp are floating point values but on version V3.8 they are integers.

closed 2024-11-24 by github-actions[bot]

blb4github, 2025-01-22:
I have checked this in version 3.9, this issue is still present.

IgorYbema, 2025-01-23:
You are correct on the R vs W. I'll change that but that is only a GUI thing. It works as it should (so they are READ by the thermostat from heishamon). That means that you needs to send over mqtt the value. 
For example on topic "thermostaat/opentherm/read/outsideTemp" send the value 10.2 and it will be set as such. So it is floating.

I just noticed a different bug and that causes the heishamon to send the wrong value as chPressure to the thermostat. I'll fix that to in a later release.

blb4github, 2025-01-23:
R vs W is indeed only GUI. 
I currently write the values via Rules to OT:

```
	?outletTemp = round(@Main_Outlet_Temp);
	?inletTemp = round(@Main_Inlet_Temp);
	?outsideTemp = #OutsideTemp;
	?dhwTemp = @DHW_Temp;
```

If I don't use the round the values are not accepted. outsideTemp and dhwTemp are integers from the HP so not a real problem for them.