heishamon/HeishaMon #557: What is the correct syntax for multiple "if - then" comparisons in Rules?

McMagellan, 2024-10-16:
I'm using version 3.8. I encountered a problem while developing my ruleset.
I don't know whether it's an error in the rulesengine or a syntax error on my part.

In the case of a conditional branch, the second condition is not calculated correctly for multiple queries if the expression is not in brackets.
If you put the condition to be calculated in the first place after the "if", the logic works correctly even without brackets.

Here is a small program that shows the problem. Under no circumstances should the query return "True".
Timer1 results in a false logic evaluation and returns "True"
Timer2 with brackets ( ) works correctly,
Timer3 outputs the correct result, but here the second value is neutral and may not have been taken into account.
Timer4 shows that the second value of the second expression is not taken into account and therefore the condition is incorrectly evaluated as "True".

```
on System#Boot then
  #Bonus = 0;
  #Periode = 160;
  settimer(1,10);
  settimer(2,11);
  settimer(3,12);
  settimer(4,13);
end

on Timer=1 then
  if #Bonus < 0 || #Bonus >= 1 + 1 then
    $Fehler = 1;
  else
    $Fehler = 0;
  end
end

on Timer=2 then
  if #Bonus < 0 || #Bonus >= (1 + 1) then
    $Fehler = 1;
  else
    $Fehler = 0;
  end
end

on Timer=3 then
  if #Bonus < 0 || #Bonus >= 1 + 0 then
    $Fehler = 1;
  else
    $Fehler = 0;
  end
end

on Timer=4 then
  if #Bonus < 0 || #Bonus >= 0 + 1 then
    $Fehler = 1;
  else
    $Fehler = 0;
  end
end
```

Attached is the console recording

```
 ==== timer=1 ====
rule #2 was executed in 1256 microseconds

>>> local variables

 0 $Fehler = 1

>>> global variables

 0 #Bonus = 0
 1 #Periode = 160

==== timer=2 ====
rule #3 was executed in 1213 microseconds

>>> local variables

 0 $Fehler = 0

>>> global variables

 0 #Bonus = 0
 1 #Periode = 160

==== timer=3 ====
rule #4 was executed in 1173 microseconds

>>> local variables

 0 $Fehler = 0

>>> global variables

 0 #Bonus = 0
 1 #Periode = 160

==== timer=4 ====
rule #5 was executed in 1209 microseconds

>>> local variables

 0 $Fehler = 1

>>> global variables

 0 #Bonus = 0
 1 #Periode = 160
```

blb4github, 2024-10-17:
I'm using brackets in all such cases, example from my ruleset:

`		if #3WayValve == 0 && (@DHW_Temp < (@DHW_Target_Temp + @DHW_Heat_Delta - 5) || (%hour == 13 && (%day == #LegionellaRunDay || @DHW_Temp < (@DHW_Target_Temp + @DHW_Heat_Delta)))) then`

closed 2024-11-23 by github-actions[bot]

CurlyMoo, 2024-12-06:
This issue should be fixed: https://github.com/IgorYbema/HeishaMon/pull/144