heishamon/HeishaMon #558: Please enable "discussions"

gvzdus, 2024-10-18:
I am absolutely delighted about my 2 Heishamon large plus Ethernet, and the stability and features.

Raising an issue for questions, or for offers for help for further engineering (I also have the Modbus-Converters, I also have the PAW-A2W-CMH-2 Cascade manager and was really unhappy with it), makes me feel bad, because every "issue" sounds a bit like a flaw or even a bug of Heishamon.

With Akkudoctors EOS project, I learnt that GitHub also has "Discussions". This is IMHO much more suitable for most of these "issues". Just check e.g. https://github.com/Akkudoktor-EOS/EOS/discussions !

So, I suggest to enable discussions for the project.

geduxas, 2024-10-18:
What topic do you think should be there?

gvzdus, 2024-10-18:
- Simple user questions, with more encouragement by the community to help
- Also "Feature requests"
- Self introduction of Heishamon users
- Hardware question (because actually Heishamon is IMHO a software project), while the hardware is really good and affordable, it is actually out of scope for the software project.

Personally, I have a bunch of both questions (e.g.: Can I (and how) use the proxy port to reverse engineer the undocumented Modbus registers of my PA-AW2-MBS-H; what is the difference between the CN-CNTs on the main board and the optional board; speculations, how Panasonics internal processing regarding compressor frequency work; the new setting in my Aquarea LT: "Quiet mode: Priority on sound or on power").

stumbaumr, 2024-10-19:
https://github.com/evcc-io/evcc/discussions is a good example on how to use Discussions in github...

gvzdus, 2024-10-21:
Thank you!
I will do my best to bring "discussions" to life!
Topics make IMHO sense...

Egyras, 2024-10-22:
Enabled discussions, free to make your topics :)

closed 2024-10-22 by Egyras

geduxas, 2024-10-25:
@gvzdus could you post more details about hardware you have? could you take internal photos? or maybe dump firmware from them?

ReinhardGruber, 2025-11-03:
https://github.com/ReinhardGruber/HeishaMonModbusTCP