heishamon/HeishaMon #559: High energy consumption by the pump in the off state.

krzbor, 2024-10-19:
I have a "K" series heat pump. When reading the "_Power_Consumption" and "_Power_Production" parameters, I was getting incorrect values of -200. However, I found in the source code (unfortunately, this is not documented in the MQTT guide) the parameters "_Power_Consumption_Extra" and "_Power_Production_Extra," which return reasonable values.

I have a question, though—when the pump is turned off, it consumes about 40W according to this data, and when it’s on but not operating, it consumes about 70W. This higher value is probably due to the operation of the water circulation pump. However, I am surprised by such a high consumption in the off state. 40W amounts to 1kWh per day!

I compared the power consumption for October. In HA, it was significantly higher than on the heat pump. However, when I subtracted 1kWh for each day, the results matched.

Is the reported 40W consumption reasonable and realistic? Are the "*_Extra" parameters reported by the pump, or are they calculated in HeishaMon?

geduxas, 2024-10-19:
I can confirm, my pump (H series) while idle consumes around same amount 
[image]

krzbor, 2024-10-19:
My pump (K series) – off state: 41W. The value comes from "Aquarea Heat Power Consumed"; topic: "panasonic_heat_pump/extra/Heat_Power_Consumption_Extra". I find it hard to believe that the new series consumes 4 times more power than the old series.

geduxas, 2024-10-19:
> My pump (K series) – off state: 41W. The value comes from "Aquarea Heat Power Consumed"; topic: "panasonic_heat_pump/extra/Heat_Power_Consumption_Extra". I find it hard to believe that the new series consumes 4 times more power than the old series.

Yes _extra topic is for new pumps K/L series, heishamon don't do any calculations, everything is fetched from pump. Also please pay attention that pump don't have any energy metering devices. All is done by calculating by pump itself.. so don't blindly trust. Second if consumption is real.. it could be from heating elements.. it could be compressor heater, it could be drain pipe heater.. and so on..

krzbor, 2024-10-19:
Maybe someone has measured the actual power consumption in the off state? If the pump doesn't have a power consumption meter, it must be calculating it somehow. It's interesting that the value shown on the pump's own graph is different from what is reported through MQTT.

stumbaumr, 2024-10-20:
I have two S0 energy meters in my power distribution cabinet that feeds into the HeishaMons S0 counters - one for the heatpump and one for the heating element of the het pump.
My 5 kW J Aquarea takes 8W in idle mode. The 200W stepping factor of the calculated power consumption is the main reason for wrong COP calculation results.
[image]

geduxas, 2024-10-20:
So ok, J an H series consumes almost same amount.. i doubth that K/L series will consume more.. it's almost same pcb. So in your @krzbor case it's something else.. it's not just pcb in idle mode or meter is incorrect

krzbor, 2024-10-20:
I believe it's the meter in the pump that's at fault. I don't think the power consumption would increase, especially by such a significant amount, in the new generation of pumps. Every manufacturer aims to reduce idle power consumption. If anyone has measurements of energy usage when the pump is off for the "K" series, please share this information.

Egyras, 2024-10-22:
first of all circulation pump consumes ~ 40W and antifrost protection for outside unit also uses +40W on top depending on outside temperature ;) Keep in mind that antifrost for outside unit kick in not constantly.

geduxas, 2024-10-22:
@Egyras for outside unit, do you talk about basepan heater or compressor crankcase?

Egyras, 2024-10-22:
> @Egyras for outside unit, do you talk about basepan heater or compressor crankcase?

original basepan heater user much more then +40W and it has 2 option for usage :)  Its just antifreeze control when circulation pump works a little bit harder, logic defined in service manual.

krzbor, 2024-10-22:
It shows 41W in the off state (the circulation pump is not working). The outside temperature is above zero.
When the circulation pump is running but there is no heating, the displayed power is 58-71W.

MiG-41, 2024-10-22:
Above 0 , but how much ? As i recall compressor heater is turning on at +3 deg....

MiG-41, 2024-10-22:
Then it is too much... 
Maybe in K counter is more precise  ( in Hor J it has resolution 0,2kW ) , but looks like still not perfect . ( Or somthing else is stealing energy). Witchout  external Wattmeter you will not be able to compare...

geduxas, 2024-10-22:
> Above 0 , but how much ? As i recall compressor heater is turning on at +3 deg....

Do crankshaft heater works all the time, or just preheats compressor before starting it? Never noticed that..

MiG-41, 2024-10-22:
All the time ( at least when compressor is not working).

closed 2024-11-27 by github-actions[bot]