heishamon/HeishaMon #563: How to use SmartGrid mode with Panasonic Aquarea L

andiwirs, 2024-10-24:
Hello everyone,

I am using a 7 kW Panasonic Aquarea L-Generation heat pump and would like to know how to utilize the SmartGrid (SG-Ready) mode.

I have installed the addititonal board CZ-NS5P and set SG setting in the installers menu, but haven't wired anything related to the SG-Ready signal yet. I have already tried to enabled the emulation of the additional board in HeishaMon, and I can also see the SmartGrid entity in Home Assistant. However, changing the settings or modifying the SmartGrid entity there doesn't seem to have any impact on the heat pump's behavior.

My goal is to simulate the SG-Ready input to increase the heat pump's output when there is a PV surplus, allowing me to regulate it based on PV generation using the SG-Ready signal.

Could you please provide guidance on how to properly set this up?

Thank you for your support!

MiG-41, 2024-10-24:
If you have CZ-NS5P you should not turn on Optional PCB emulation in Heishamon !

This is how to do it, you have to shortcut/open proper terminals on CZ-NS5P:
[image]

andiwirs, 2024-10-24:
Thanks for your answer. Does this mean that there is no software emulation/simulation? So I really have to wire this up? Thanks :)

geduxas, 2024-10-24:
> Thanks for your answer. Does this mean that there is no software emulation/simulation? So I really have to wire this up? Thanks :)

If you have expansion card installed when there is nothing to emulate.. emulation and real card can't operate in same time.

In emulation you have virtual switches, in real board real contacts..

gvzdus, 2024-10-26:
I have the cascade manager, which claims to have PV optimization. What it simply does is increasing the target temperature (for heating / DHW) by +5K. So: You can do the same with Heishamon already now.
"SG ready" is the "universal stupid" way, you do have the smart way with Heishamon in your pocket :-)

geduxas, 2024-10-26:
Yes. With heishamon you have much more control.. as for SG contact"s i think you have same cascade functionality.. for example you set sg contact to 150% of heating, so i think it should rise same +5K as controller do.. (never tested) 

And credit's going to @edterbak who made amazing NodeRed integration/automation which have DHW control logic tied to solar production.
https://github.com/edterbak/NodeRed_Heishamon_control

closed 2024-12-01 by github-actions[bot]