heishamon/HeishaMon #564: Auto-discovery doesn't work with HeishaMon module

Maystero14, 2024-10-24:
I have been trying to integrate HeishaMon with Home Assistant for some time to manage my Panasonic heat pump. I have searched for help on discussion forums. To no avail. I currently have an ESP8266 module running with HeishaMon firmware 3.8. MQTT server configured and running. Ping communication between devices works. And Home Assistant does not detect entities from the HeishaMon device I also installed packages from HACS Heishamon, but that didn't help either. MQTT Explorer sees mqtt packets, so the system seems to be working correctly.

[image]
[image]

Thanks in advance for every help.

geduxas, 2024-10-24:
There is no autodiscovery compatible with HA taip n heishamon, earlier we used manual configuration, now it's outdated a lot.. i suggest you to use Heishamon integration from HACS.

Maystero14, 2024-10-24:
> Sorry, didn't noticed that you are trying with integration, I don't see topics in your mqtt, do you get any values in heishamon web page?

You means from ESP8266?

Maystero14, 2024-10-24:
Fixed

closed 2024-10-24 by Maystero14