heishamon/HeishaMon #568: Home Assistant Configuration for K/L Version

krzbor, 2024-10-29:
A new YAML file for Home Assistant has just been released. It is worth preparing a version for the K/L series pumps. These pumps report information about consumed and generated energy differently.

The following entries need to be modified:
    #TOP15 - Heat power produced XTOP3
    - name: Aquarea Heat Power Produced
      state_topic: "panasonic_heat_pump/extra/Heat_Power_Production_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

    #TOP16 - Heat power consumed XTOP0
    - name: Aquarea Heat Power Consumed
      state_topic: "panasonic_heat_pump/extra/Heat_Power_Consumption_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

    #TOP38 - Cool power produced XTOP4
    - name: Aquarea Cool Power Produced
      state_topic: "panasonic_heat_pump/extra/Cool_Power_Production_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

    #TOP39 - Cool power consumed XTOP1
    - name: Aquarea Cool Power Consumed
      state_topic: "panasonic_heat_pump/extra/Cool_Power_Consumption_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

    #TOP40 - DHW power produced XTOP5
    - name: Aquarea DHW Power Produced
      state_topic: "panasonic_heat_pump/extra/DHW_Power_Production_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

    #TOP41 - DHW power consumed XTOP2
    - name: Aquarea DHW Power Consumed
      state_topic: "panasonic_heat_pump/extra/DHW_Power_Consumption_Extra"
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      force_update: true

geduxas, 2024-10-29:
Make change's and PR. I doubt anyone still using manual setup.. but if anyone is required, so why not.

krzbor, 2024-10-29:
And which configuration in HA should it use?

geduxas, 2024-10-29:
Heishamon integration over HACS, https://github.com/kamaradclimber/heishamon-homeassistant

krzbor, 2024-10-29:
This requires installing HACS, and not everyone wants that. By editing YAML, I was able to remove items from dropdown lists, translate dropdown options into the native language, add a "utility_meter," round results to whole numbers, etc.

geduxas, 2024-10-29:
Ok, it's up to you :) everyone is able to choose. So if you think it should be in yaml, so please make changes and just PR code..

krzbor, 2024-10-29:
I described the changes that need to be made in the YAML file for K/L pumps. It took me some time to find the issue with incorrect power information. The MQTT HeishaMon documentation does not include this information. Fortunately, there is the source code.

closed 2024-12-04 by github-actions[bot]