heishamon/HeishaMon #569: Request: emulation of inside PCB

joep2000, 2024-10-31:
The gen M doesn't have the inside PCB like previous models. Only connections possible are 3-way valve, tank sensor, TAW1. (Unless you purchase the controlbox, hydraubox or all-in-one).
It would be nice that HeishaMon can emulate the inside PCB, same as the optional board (connection of external thermostat, extra pump, buffer sensor/heater, etc)

MiG-41, 2024-11-01:
If it doesn't have anything extra , how to emulate it ?  Probably it is not accepting that commands from Optional PCB. 
First it would be nice to somone have this HP , confirm ,that standard CZ-TAW1 works with it , do some sniffing , ect.

joep2000, 2024-11-01:
@MiG-41 
Some more background. I own a12kW gen M (WH-WXG12ME8 with Remote Control CZ-RTW2TAW1C) that is connected with the Heishamon large PBC and the TAW1. From fist observations I can confirm that both are working.

However the gen M is a bit of a different beast. The 'cheap' version doesn't come with the indoor PCB as described above. Just the pump and the remote control. Only connections possible are for DHW tank option: 3way valve and tank sensor. So the bare minimum.
Pay 1200€ extra and you'll get the 'control box'. This box communicates with the pump and includes the indoor PCB: all of the connections you'll normally find in a Pana pump.

Other options are the 'Bi-block' which includes a heating element. 
Or the All-In-One, which includes all ;-) (DHW, heating element, controls). 
With these indoor options for the gen M, you have the optional PCB as well.

So, emulating the indoor PCB would help spectacularly (for saving space inside and €€€)

MiG-41, 2024-11-01:
Maybe first you could tell ,how this 'control box' looks like , how it would be connected to mainboard of M series ? Maybe some pictures of M PCB board ? ( i know that this could be found , there are also probably service manual for M series already , but you have to also understand ,that users with are own HP previous generations not following the changes with such entusiasm as before :) )

joep2000, 2024-11-01:
Sure, I understand that the M has just launched, and it is a T-CAP. However, I think Panasonic is moving in this direction to decrease the price of the main unit. So consider this work for future benefit (and my benefit for now).

Indoor units communicate with a 2 wire interface (I have no idea which one, but probably Modbus or some variation). TAW1 connection is also available.

So let me help myself figuring out how to tackle this. Help appriciated!

MiG-41, 2024-11-01:
2 wire interface could be similar to one , with communicates with display controller...  For with we are also not able to influence ( you can't set a data ,clock , brithness , edit schedules ect).

joep2000, 2024-11-01:
unit inside:
[image]

in a sealed box (R290) on top all power electronics:
[image]
[image]
[image]

Service manual is pretty thin though (took out a few pages to shrink below 25Mb. ). 
Also, this docuement is focused on the All-In-One:
[(EN) Panasonic M-gen service manual.pdf](https://github.com/user-attachments/files/17598763/EN.Panasonic.M-gen.service.manual.pdf)

MiG-41, 2024-11-01:
From this:
[image]

You can see , that communication beetween ( here also called ) OPTIONAL is via second CN-CNT ( or first ,since second is called CN-CNT2 and it is for Network adabter - assumption CZ-TAW1).

So only thing with is needed is to sniff comunication beetween this two devices ( but to sniff you need them booth).

geduxas, 2024-11-01:
It's everything same as in previous versions, and with k/l boards.. i think even same optional(expansion) board. 
As for anode, yes it have active anode in boiler, previous boiler's don't even have any anode. 

Difference from H/J series that some element's are moved from expansion board to main board. So everything what is on mainboard can't be emulated.

geduxas, 2024-11-01:
> unit inside: ![20240816_125635](https://private-user-images.githubusercontent.com/32467222/382251774-02b35c95-de6f-4d2b-a11f-e48f4310fa62.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0ODUxMzksIm5iZiI6MTczMDQ4NDgzOSwicGF0aCI6Ii8zMjQ2NzIyMi8zODIyNTE3NzQtMDJiMzVjOTUtZGU2Zi00ZDJiLWExMWYtZTQ4ZjQzMTBmYTYyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDE4MTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1OTg0NTk5MTY0ZDY4Y2E3YzVjZjA5NDU4OGMwYmNlNDM5M2UyOGY2YTcyM2RjN2Q2YTQwNDE0NjcyMTc3MjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.arRdZ4JTQXGaQvKuHGOKKJRm05A8oxKZoq0hWxqqtTE)
> 
> in a sealed box (R290) on top all power electronics: ![20240817_123453](https://private-user-images.githubusercontent.com/32467222/382251782-d2a20835-9873-4624-917d-fed59feaa5cd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0ODUxMzksIm5iZiI6MTczMDQ4NDgzOSwicGF0aCI6Ii8zMjQ2NzIyMi8zODIyNTE3ODItZDJhMjA4MzUtOTg3My00NjI0LTkxN2QtZmVkNTlmZWFhNWNkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDE4MTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyYjc2MTkwOTczYmY5ZDc5ZGQ5YTk4ZDVjNTc3ZTZhNGI2N2JhZDM5NzI4ODE4YTRiYWY3ZWFjZTI2MDM2MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wdOjPSzzRJgzDH5KXmgglTzazOzz1GvQoPzayw-6kig) ![20240817_123457](https://private-user-images.githubusercontent.com/32467222/382251785-97e3569c-34b1-46f6-99f0-e7811cec771f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0ODUxMzksIm5iZiI6MTczMDQ4NDgzOSwicGF0aCI6Ii8zMjQ2NzIyMi8zODIyNTE3ODUtOTdlMzU2OWMtMzRiMS00NmY2LTk5ZjAtZTc4MTFjZWM3NzFmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDE4MTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmMzdkY2U1MmIxYzg5MzViMTA0YmI0M2RjZTI3ZGU0NWYzOWM5MTQwMzQ2ZDU3M2ZhMjZkZDYzYjI2ZGQ1ZDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jR6O6ulL-tZ2hfBqDD_Ef7-hVQM1LY_0cNFOrWOKqjc) ![20240817_123501](https://private-user-images.githubusercontent.com/32467222/382251790-7e0c6348-b1cc-48bb-8319-cacf5142ac89.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0ODUxMzksIm5iZiI6MTczMDQ4NDgzOSwicGF0aCI6Ii8zMjQ2NzIyMi8zODIyNTE3OTAtN2UwYzYzNDgtYjFjYy00OGJiLTgzMTktY2FjZjUxNDJhYzg5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDE4MTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI2NDNkNmE0ZjhmYWEzMWM1NTlmMzljNjFmYTY1ZDcxYTM3M2RhOTBmZjEwNTcwMDFiYjUxNzdkNWJjM2VhYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.X2WhgoHSGEZsFLH4myJCoGluvsHzDuKr2EBNbLA_Uoc)
> 
> Service manual is pretty thin though (took out a few pages to shrink below 25Mb. ). Also, this docuement is focused on the All-In-One: [(EN) Panasonic M-gen service manual.pdf](https://github.com/user-attachments/files/17598763/EN.Panasonic.M-gen.service.manual.pdf)

It's just outdoor unit boards, you're missing main board, could it be behind screws where paper attached?

geduxas, 2024-11-01:
According to service manual it's hydro-split unit, so main board in inside unit, and it's looking same as k/l series.. do we are missing something? Do you have inside unit? Or just outside?

joep2000, 2024-11-02:
@MiG-41 You are describing the indoor unit
@geduxas yes you are missing something :-)

chapter 9.2 (p79 of service manual) describes the indoor unit. Only connections are:
-tank sensor (CN-TH)
-Remote controller (CN-RC)
-TAW1 (and heishamon) (CN-CNT)
-indoor unit (CN-DATA)
[image]

AND:
-3way valve (CN-V3)
-boiler contact (CN-CONCTACT)
[image]

Both main PCB and Optional PCB are in the indoor units. Which are:
- Control Box (WH-CME8)
- Bi-Bloc T-CAP (WH-SDC0316M9E8)
- Al-In-One T-CAP (WH-ADC0316M9E82) (the service manual is for this one, so, hence the confusion above)

I only have the outdoor unit. Misdirected by the sales people, it only has the few connections stated above. I hope Heishamon can eventualy replace the indoor units, since they start from 1200€.

geduxas, 2024-11-02:
Oh, so i thought right, yes ODU and IDU are sold separately all time, just didn't though it could run standalone. But i have som thought's about hydro-split units, because it doesn't seemed something ok with that.. now i understand.. ok, so you don't have inside unit.. but your almost have minim what is needed for pump to operate.. 

So what you're missing? What still needs to be emulated?

geduxas, 2024-11-02:
> It would be nice that HeishaMon can emulate the inside PCB, same as the optional board (connection of external thermostat, extra pump, buffer sensor/heater, etc)

External thermostat it's in main pcb.. it could be possible or impossible to emulate i think it will require more data.. somewhere should be all pump logic, naturally it's on inside unit (but questionable).

Extra pump (you are talking about pump per zone?) It's emulated
Buffer sensor also emulated.. 

Have you tested emulated setting while connecting to cn-cnt? Or you even don't have possibility to enable expansion board from RC?

What about jumperwire which is required for stand alone operation, have you tried to remove it?

joep2000, 2024-11-03:
I haven't tried removing the jumperwire. I guess that the HP then expects it has a main board, but it hasn't. So probably won't do anything except throw an error...

I cannot select the Optional board in the settings, so I guess that is out of the question for emulating all the extra's I'd like.

- Extra pump: I'm running with a parrallel-buffer. Extra pump is just to switch with the main pump. This can be done with rules in Heishamon, so not really important at this point.
- Buffer sensor is emulated in Optional Board? Can't see it in the MQTT topic list, anyway, at the moment can't do that since I dont have the optional board
- SG, would be nice

geduxas, 2024-11-03:
Ok, so yes, i think by removing jumperwire it will forward logic to main board.. so it won't work in this situation. It will require additional investigation on port and protocol.

For that we needed volunteers with full setup and dataloger, to analyze traffic..

MiG-41, 2024-11-03:
> * Extra pump: I'm running with a parrallel-buffer. Extra pump is just to switch with the main pump. This can be done with rules in Heishamon, so not really important at this point.

Extra Pump also not working ,when DHW is boiling... but that is deatail.

> * Buffer sensor is emulated in Optional Board? Can't see it in the MQTT topic list, anyway, at the moment can't do that since I dont have the optional board

Yes ,on optional PCB there is a Buffer sensor to be placed on the top of buffer. That is one of least reverse enginered function , since no one is using this , and next case is that it apear also next delta  ( additional temperature ) with is higher in buffer....  
With is silly and loosing CAP , since HP is heating additional temp. with is storaging in buffer , and for floor is going cooler water...

geduxas, 2024-11-03:
@MiG-41 i have buffer, with 2 zones.. so yes, buffer target temperature is highest zone target temperature plus buffer delta. Original 50L buffer comes with temperature sensor in midle.

MiG-41, 2024-11-03:
But you are using it ? You have real Optional PCB ? If yes , you could investigate for with Byte09 in OptionalPCB command is ( i belive it is something related to the buffer , but never had a chance to check it).

geduxas, 2024-11-04:
Yes i have everything needed for 2 zones. Ill do my best to investigate byte 09 in my setup.

geduxas, 2024-12-05:
month passed :D and i still not started anything

closed 2025-01-12 by github-actions[bot]