heishamon/HeishaMon #573: Can't pass identical variables as parameters to function in Rules

stundenblume, 2024-11-06:
Hi,
first: Thank you so much for this very useful software. I really enjoy your work!

Seems to be a small bug wich is easily bypassable. 

Problem: This don't work: `action(#foo, #foo);` After saving the rule, the website loads a long time and will end with an "Error: Connection Lost". After reloading the website the rule is unchanged.
This works: `action(#foo, #bar);`
~~~
on System#Boot then
  #foo = 0;
  #bar = #foo;
  setTimer(1, 10);
end

on action($x, $y) then
 $z = $x + $y;
end

on timer=1 then
  action(#foo, #foo);
end
~~~

Testet on the small and big PCB.
Software Version: 3.8

CurlyMoo, 2024-12-07:
Should be fixed in https://github.com/IgorYbema/HeishaMon/pull/144

stundenblume, 2024-12-07:
Thank you!
I will wait for the next release and try it.

closed 2025-01-13 by github-actions[bot]