heishamon/HeishaMon #575: Heishamon V5 cannot be put into operation. (Firmware 3.8)

McMagellan, 2024-11-07:
My Heishamon V5 board arrived today and I tried in vain to put it into operation.
The connection to the access point works immediately and I can see the current data in the console.
Without a connection to the HP, I tried entering the WiFi data in the settings. As soon as I execute Save, there is no response and the "savesettings" window times out. However, a factory reset is carried out correctly.

Loading a rule is partially possible.

This code is accepted:
```
on System#Boot then
  $Test1 = 55;
end
```

This code times out and is not accepted.
```
on System#Boot then
  $Test1 = "55";
end
```

I recorded more information on my old Heishamon V4 at the debug port. But I can't find any connections for debugging on the new V5 board.

@IgorYbema :  How is it possible to record debug data on V5 and what does “Debug log USB” mean?
I decided not to load any other firmware for the time being because of this unsafe behavior.

geduxas, 2024-11-07:
Debug log usb could mean that usb will work as serial port, and maybe heishamon will output debug to it?

Just geasing.

McMagellan, 2024-11-07:
I have Heishamon V5 lying next to the PC and connected the USB socket to a USB port on the PC with a cable. It would now be ideal if the PC recognized an interface that I could then read in using Putty Serial. Unfortunately that doesn't happen. I think that the USB socket can only be used for 5V power supply.

Unfortunately it is not possible to enter WiFi data, which means Heishamon cannot be used. Likewise, no MQTT connection is possible.

geduxas, 2024-11-07:
According source code yes usb should be tied to serial interface. And after enabling usb debugging, i think it needs to be restarted. Also don't know how hardware is implemented, does it have some serial driver, or usb is directly tied to esp32? Maybe you could share detailed photos? It could be that your need some extra drivers..

McMagellan, 2024-11-08:
I ran further tests to find out what the problems were.

1) Settings cannot be changed.
I connected to the Heishamon V5 access point with my desktop and later with my laptop and was also able to establish the connection via http://192.168.4.1/.
Saving the settings failed and resulted in a page timeout. Both PCs use Firefox as a browser. Then I connected to my cell phone and was able to change the settings without any problems. Then I tested Firefox in private mode from the desktop and it now worked here too. 
So what is different about private mode compared to normal mode?

2) Rule set is not accepted.
I couldn't load my CoPilot ruleset with 33 rules exactly as I successfully loaded it into Heishamon4 into Heishamon V5 because it wasn't accepted. 
Research has shown that the problem lies in the assignment of a variable type String when " is used as a separator. 
See test rule in the post above.
If I now replace the separator " with the separator ' the ruleset is also accepted in V5. 

Old Version failed:
```
on System#Boot then
  $Version = "6.22 McM";
  $Datum = "30.10.2024";
  settimer(11,10);
end
```

New Version accepted:
```
on System#Boot then
  $Version = '6.22 McM';
  $Datum = '30.10.2024';
  settimer(11,10);
end
```

The still open issue is how I can connect to the debug channel.

McMagellan, 2024-11-12:
Addendum: Today I connected HeishamonV5 to my laptop again with a different USB3 cable. 
A serial Putty connection was established through which I can record the debug information just like with HeishamonV4. Great job.
The problem was with me and a USB cable that had no data connection but was only suitable for charging.

Now all I'm missing is the information on how the relays can be controlled

geduxas, 2024-11-12:
Could it be SetGPIO16State ?

McMagellan, 2024-11-12:
I don't know what you mean by that, I'm not an expert, just a hobby rules programmer.
In contrast to the console window, the debug channel has several pieces of information that are important to me.
For example errorinfornation while parsing/loading a ruleset and TOP trigger Info.
I simply connected the USB interface of the Heishamon V5 to my laptop and the port was immediately recognized as COM5 with a standard USB driver (needs nothing to install).

However, the 5V power of the laptop was not enough to function correctly, which is why I connected an extra 5V power supply to the 1wire port. I use an RS485 extension and definitely need a local power supply for the Heishamon. 
I also wanted a galvanic isolation from the Jeisha for security reasons.

Debug via USB is activated in the settings.
With Putty via serial I can then query the interface and record it even for long time. (whole days)
Here is a short log file for information.

[1211aputty.log](https://github.com/user-attachments/files/17723107/1211aputty.log)

geduxas, 2024-11-12:
This board is new, also i am waiting for delivery also, interested to test it ;) and yes.. there is no documentation (i think) but there was topic about it look here #579

McMagellan, 2024-11-12:
Now I understand what you mean by "Could it be SetGPIO16State ?" You meant the control of the relays.
**I need a working example.** 
The relays are definitely connected to ports on the ESP, but I don't know how to control them via a URL command and which ones they are. I imagine that they can also be addressed from within rules, for example like a local variable.

I also saw your linked issue #579 and posted a post as a question.

I am enthusiastic about HeishamonV5 after there were initial problems that I was always able to overcome somehow. Now I am also connected to the Panasonic Service Cloud in parallel through the proxy mode, which further enhances the HP for me.

geduxas, 2024-11-12:
From source code i see it's possible only from mqtt. I think it's also easily port to web and Into rules.. for now it's just basic functionality. Also don't see any states. Just physical LED indication.  So setgpio16 command is for something different maybe for opentherm.

McMagellan, 2024-11-12:
On the Heishamon V5 board there is a 2-pin terminal connection at the bottom left. This is labeled "OT". The label is not visible in this picture. Perhaps this is the connection for a bus system that has not yet been published and the command: SetGPIO16State belongs to it. 
I don't know OpenTherm, what is it good for?

[image]

closed 2024-12-13 by McMagellan