heishamon/HeishaMon #577: Power and cop after update to 3.62

anno006, 2024-11-09:
The sensors will not update anymore after update 3.62

heat_consumption
Heat_production
COP

domoticz useing the plugin on 2024.7 16178

Tailer67, 2024-11-10:
Have you noticed that?

https://github.com/Egyras/HeishaMon/releases/tag/v3.2

anno006, 2024-11-10:
??? i don't have Opentherm and don't use it. My post is about the energy meters and the COP sensor (not the s-0 meters) that won't get an update anymore from heishamon. after updating to 3.6 and .8 So somewhere between 3.2 and 3.6-.8 something is broken!

MiG-41, 2024-11-10:
Next hint:

`Rename 'Energy' to 'Power' for the heat/cool/dhw consumption and production topics`

munion-2024, 2024-11-11:
I've another problem since I've updated to 3.8, there are a lot of error messages in the log:

2024-11-11 01:09:02.718  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 613, in onMessage
2024-11-11 01:09:02.718  Error: Heisha:     _plugin.onMessage(Connection, Data)
2024-11-11 01:09:02.718  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 293, in onMessage
2024-11-11 01:09:02.718  Error: Heisha:     self.mqttClient.onMessage(Connection, Data)
2024-11-11 01:09:02.718  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/mqtt.py", line 157, in onMessage
2024-11-11 01:09:02.718  Error: Heisha:     payload =  Data['Payload'].decode('utf8') if 'Payload' in Data else ''
2024-11-11 01:09:02.718  Error: Heisha: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 1: invalid continuation byte
2024-11-11 01:09:07.728  Error: Heisha: Call to function 'onMessage' failed, exception details:
2024-11-11 01:09:07.728  Error: Heisha: Traceback (most recent call last):
2024-11-11 01:09:07.728  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 613, in onMessage
2024-11-11 01:09:07.729  Error: Heisha:     _plugin.onMessage(Connection, Data)
2024-11-11 01:09:07.729  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 293, in onMessage
2024-11-11 01:09:07.729  Error: Heisha:     self.mqttClient.onMessage(Connection, Data)
2024-11-11 01:09:07.729  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/mqtt.py", line 157, in onMessage
2024-11-11 01:09:07.729  Error: Heisha:     payload =  Data['Payload'].decode('utf8') if 'Payload' in Data else ''
2024-11-11 01:09:07.729  Error: Heisha: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 1: invalid continuation byte
2024-11-11 01:09:12.363  RFXCom: Temp + Humidity (Temp Corral)
2024-11-11 01:09:12.623  Status: dzVents: Info: Handling events for: "Temp Corral", value: "10.6;80;3"
2024-11-11 01:09:12.623  Status: dzVents: Info: ------ Start internal script: D Envio a Wunderground: Device: "Temp Corral (RFXCom)", Index: 794
2024-11-11 01:09:12.624  Status: dzVents: Info: ------ Finished D Envio a Wunderground
2024-11-11 01:09:12.625  Status: EventSystem: Script event triggered: /home/pi/domoticz/dzVents/runtime/dzVents.lua
2024-11-11 01:09:12.645  Status: EventSystem: Script event triggered: D Guardo Temp max-min
2024-11-11 01:09:12.738  Error: Heisha: Call to function 'onMessage' failed, exception details:
2024-11-11 01:09:12.739  Error: Heisha: Traceback (most recent call last):
2024-11-11 01:09:12.739  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 613, in onMessage
2024-11-11 01:09:12.740  Error: Heisha:     _plugin.onMessage(Connection, Data)
2024-11-11 01:09:12.740  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line 293, in onMessage
2024-11-11 01:09:12.740  Error: Heisha:     self.mqttClient.onMessage(Connection, Data)
2024-11-11 01:09:12.740  Error: Heisha:   File "/home/pi/domoticz/plugins/HeishamonMQTT/mqtt.py", line 157, in onMessage
2024-11-11 01:09:12.740  Error: Heisha:     payload =  Data['Payload'].decode('utf8') if 'Payload' in Data else ''
2024-11-11 01:09:12.740  Error: Heisha: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 1: invalid continuation byte

munion-2024, 2024-11-11:
Anno006, you need to create new devices on Domoticz, this is what I've done:

- Replace "Energy" with "Power" in your Heishamon plugin.py file. I'm running Domoticz on a raspberry
sed -i 's/Energy/Power/g' /home/pi/domoticz/plugins/HeishamonMQTT/plugin.py

- In Domoticz, setup, Settings click "Allow for 5 minutes"

- In Domoticz, Setup, Hardware find your Heishamon MQTT hardware, click on int and after that click on Replace button

After a few seconds you will see nine new devices, Power consumption, Power Producction and COP for Heat, Cool and DHW.

Regards

munion-2024, 2024-11-11:
Anno006, you need to create new devices on Domoticz, this is what I've done:

    Replace "Energy" with "Power" in your Heishamon plugin.py file. I'm running Domoticz on a raspberry
    sed -i 's/Energy/Power/g' /home/pi/domoticz/plugins/HeishamonMQTT/plugin.py

    In Domoticz, setup, Settings click "Allow for 5 minutes"

    In Domoticz, Setup, Hardware find your Heishamon MQTT hardware, click on int and after that click on Replace button

After a few seconds you will see nine new devices, Power consumption, Power Producction and COP for Heat, Cool and DHW.

Regards
[image]

anno006, 2024-11-12:
> I've another problem since I've updated to 3.8, there are a lot of error messages in the log:
>
> 2024-11-11 01:09:02.718 Error: Heisha: File "/home/pi/domoticz/plugins/HeishamonMQTT/plugin.py", line [...]

This i did from the start when i did run on 3.0 so this problem is really old!

munion-2024, 2024-11-12:
Sorry, I didn't mean to offend, I just wanted to help without knowing your abilities. Maybe this will help someone ignorant like me.

Best regards

geduxas, 2024-11-12:
> another hint, what was wrong with the word Energy? also why does it absolutly need to change into Power? Why didn't you all do it youself in the domoticz heishamon plug here on github?
> 
> And last but not least, it is stupid! i do have a lot of luck that i run this domoticz on ubuntu desktop 20.04.6 and that i know where to look!

Energy, and power is 2 different things by meaning, it was wrong from beginning, and in some point it was decided to change. 
As for domoticz, i think there was no one interested to change... Heishamon is open to modify as you like.. so if you made some changes fill free to make changes.

closed 2024-12-19 by github-actions[bot]