heishamon/HeishaMon #578: Problem with Heat_Power_Production and Heat_Power_Consumption allways -200

alf1964m, 2024-11-10:
Hello, I have a problem with my Heishamon large board. The display for consumed electricity and generated heating power is always set to -200 watts. The firmware version is 3.8 The board is two weeks old. 
[Heisha monitor.pdf](https://github.com/user-attachments/files/17692003/Heisha.monitor.pdf)
[image]

In the attachment I send you a copy of the settings and the console. Can you please tell me if there is something wrong, or if the board is defective? Thank you very much. Best regards Elmar

`Sun Nov 10 13:42:45 2024 (442867952): Previous read data attempt failed due to timeout!
Sun Nov 10 13:42:45 2024 (442867953): Received 0 bytes data
Sun Nov 10 13:42:45 2024 (442867954): Sending optional PCB data
Sun Nov 10 13:42:45 2024 (442867955): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:42:47 2024 (442869957): Previous read data attempt failed due to timeout!
Sun Nov 10 13:42:47 2024 (442869958): Received 0 bytes data
Sun Nov 10 13:42:47 2024 (442869959): Sending optional PCB data
Sun Nov 10 13:42:47 2024 (442869960): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:42:49 2024 (442871962): Previous read data attempt failed due to timeout!
Sun Nov 10 13:42:49 2024 (442871962): Received 0 bytes data
Sun Nov 10 13:42:49 2024 (442871963): Sending optional PCB data
Sun Nov 10 13:42:49 2024 (442871964): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:42:51 2024 (442873966): Previous read data attempt failed due to timeout!

Sun Nov 10 13:43:20 2024 (442902625): Heishamon stats: Uptime: 5 days 3 hours 1 minute 42 seconds ## Free memory: 68% ## Free PSRAM: 2070100 bytes ## Free heap: 234236 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 2 ## Correct data: 4.31% Rules active: 0
Sun Nov 10 13:43:20 2024 (442902628): Requesting new panasonic data
Sun Nov 10 13:43:20 2024 (442902629): Already sending data. Buffering this send request
Sun Nov 10 13:43:20 2024 (442902629): Requesting new panasonic extra data
Sun Nov 10 13:43:20 2024 (442902630): Already sending data. Buffering this send request
Sun Nov 10 13:43:21 2024 (442904032): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:21 2024 (442904032): Received 0 bytes data
Sun Nov 10 13:43:21 2024 (442904033): Sending command from buffer
Sun Nov 10 13:43:21 2024 (442904034): sent bytes: 111 including checksum value: 18 
Sun Nov 10 13:43:22 2024 (442904441): Received 203 bytes data
Sun Nov 10 13:43:22 2024 (442904442): Checksum and header received ok!
Sun Nov 10 13:43:22 2024 (442904442): received TOP1 Pump_Flow: 13.74
Sun Nov 10 13:43:22 2024 (442904446): received TOP51 Inside_Pipe_Temp: 35
Sun Nov 10 13:43:22 2024 (442904449): received TOP117 Economizer_Outlet_Temp: 118
Sun Nov 10 13:43:22 2024 (442904454): Sending command from buffer
Sun Nov 10 13:43:22 2024 (442904455): sent bytes: 111 including checksum value: 1 
Sun Nov 10 13:43:24 2024 (442906457): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:24 2024 (442906457): Received 0 bytes data
Sun Nov 10 13:43:24 2024 (442906458): Sending optional PCB data
Sun Nov 10 13:43:24 2024 (442906459): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:26 2024 (442908461): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:26 2024 (442908461): Received 0 bytes data
Sun Nov 10 13:43:26 2024 (442908462): Sending optional PCB data
Sun Nov 10 13:43:26 2024 (442908463): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:28 2024 (442910465): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:28 2024 (442910465): Received 0 bytes data
Sun Nov 10 13:43:28 2024 (442910466): Sending optional PCB data
Sun Nov 10 13:43:28 2024 (442910467): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:30 2024 (442912469): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:30 2024 (442912469): Received 0 bytes data
Sun Nov 10 13:43:30 2024 (442912470): Sending optional PCB data
Sun Nov 10 13:43:30 2024 (442912471): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:32 2024 (442914473): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:32 2024 (442914473): Received 0 bytes data
Sun Nov 10 13:43:32 2024 (442914474): Sending optional PCB data
Sun Nov 10 13:43:32 2024 (442914475): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:34 2024 (442916477): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:34 2024 (442916477): Received 0 bytes data
Sun Nov 10 13:43:34 2024 (442916478): Sending optional PCB data
Sun Nov 10 13:43:34 2024 (442916479): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:36 2024 (442918481): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:36 2024 (442918481): Received 0 bytes data
Sun Nov 10 13:43:36 2024 (442918482): Sending optional PCB data
Sun Nov 10 13:43:36 2024 (442918483): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:38 2024 (442920485): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:38 2024 (442920485): Received 0 bytes data
Sun Nov 10 13:43:38 2024 (442920486): Sending optional PCB data
Sun Nov 10 13:43:38 2024 (442920487): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:43:40 2024 (442922489): Previous read data attempt failed due to timeout!
Sun Nov 10 13:43:40 2024 (442922489): Received 0 bytes data
[... 173 lines cut ...]
Sun Nov 10 13:44:06 2024 (442949256): received TOP112 Z2_Sensor_Settings: 0
Sun Nov 10 13:44:06 2024 (442949258): received TOP113 Buffer_Tank_Delta: 8
Sun Nov 10 13:44:06 2024 (442949259): received TOP114 External_Pad_Heater: 0
Sun Nov 10 13:44:06 2024 (442949261): received TOP115 Water_Pressure: 1.34
Sun Nov 10 13:44:06 2024 (442949262): received TOP116 Second_Inlet_Temp: 38
Sun Nov 10 13:44:06 2024 (442949264): received TOP117 Economizer_Outlet_Temp: 119
Sun Nov 10 13:44:06 2024 (442949266): received TOP118 Second_Room_Thermostat_Temp: 127
Sun Nov 10 13:44:06 2024 (442949268): Sending command from buffer
Sun Nov 10 13:44:06 2024 (442949268): sent bytes: 111 including checksum value: 1 
Sun Nov 10 13:44:08 2024 (442951271): Previous read data attempt failed due to timeout!
Sun Nov 10 13:44:08 2024 (442951271): Received 0 bytes data
Sun Nov 10 13:44:08 2024 (442951272): Sending optional PCB data
Sun Nov 10 13:44:08 2024 (442951273): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:44:10 2024 (442953275): Previous read data attempt failed due to timeout!
Sun Nov 10 13:44:10 2024 (442953275): Received 0 bytes data
Sun Nov 10 13:44:10 2024 (442953276): Sending optional PCB data
Sun Nov 10 13:44:10 2024 (442953277): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:44:12 2024 (442955279): Previous read data attempt failed due to timeout!
Sun Nov 10 13:44:12 2024 (442955279): Received 0 bytes data
Sun Nov 10 13:44:12 2024 (442955280): Sending optional PCB data
Sun Nov 10 13:44:12 2024 (442955281): sent bytes: 20 including checksum value: 164 
Sun Nov 10 13:44:14 2024 (442957286): Previous read data attempt failed due to timeout!
Sun Nov 10 13:44:14 2024 (442957287): Received 0 bytes data

`

MiG-41, 2024-11-10:
It is becouse Heishamon detects , that you have HP newer generation  , and proper values are in positions XTOP ) with you have -1W anyway....).
Please check,if in this XTOP topics values are more accurate during HP working.

alf1964m, 2024-11-10:
Hello i hope you can see more infos.
`Sun Nov 10 20:08:32 2024 (466014254): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:32 2024 (466014256): Received 0 bytes data
Sun Nov 10 20:08:32 2024 (466014257): Sending optional PCB data
Sun Nov 10 20:08:32 2024 (466014258): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:32 2024 (466014258): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:34 2024 (466016261): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:34 2024 (466016261): Received 0 bytes data
Sun Nov 10 20:08:34 2024 (466016262): Sending optional PCB data
Sun Nov 10 20:08:34 2024 (466016263): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:34 2024 (466016264): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:36 2024 (466018266): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:36 2024 (466018266): Received 0 bytes data
Sun Nov 10 20:08:36 2024 (466018267): Sending optional PCB data
Sun Nov 10 20:08:36 2024 (466018268): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:36 2024 (466018269): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:38 2024 (466020271): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:38 2024 (466020271): Received 0 bytes data
Sun Nov 10 20:08:38 2024 (466020272): Sending optional PCB data
Sun Nov 10 20:08:38 2024 (466020273): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:38 2024 (466020274): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:40 2024 (466022276): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:40 2024 (466022276): Received 0 bytes data
Sun Nov 10 20:08:40 2024 (466022277): Sending optional PCB data
Sun Nov 10 20:08:40 2024 (466022278): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:40 2024 (466022279): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:42 2024 (466024281): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:42 2024 (466024281): Received 0 bytes data
Sun Nov 10 20:08:42 2024 (466024282): Sending optional PCB data
Sun Nov 10 20:08:42 2024 (466024283): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:42 2024 (466024284): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:44 2024 (466026286): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:44 2024 (466026286): Received 0 bytes data
Sun Nov 10 20:08:44 2024 (466026287): Sending optional PCB data
Sun Nov 10 20:08:44 2024 (466026288): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:44 2024 (466026289): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:46 2024 (466028291): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:46 2024 (466028291): Received 0 bytes data
Sun Nov 10 20:08:46 2024 (466028292): Sending optional PCB data
Sun Nov 10 20:08:46 2024 (466028293): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:46 2024 (466028294): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:48 2024 (466030296): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:48 2024 (466030296): Received 0 bytes data
Sun Nov 10 20:08:48 2024 (466030297): Sending optional PCB data
Sun Nov 10 20:08:48 2024 (466030298): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:48 2024 (466030299): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:50 2024 (466032301): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:50 2024 (466032301): Received 0 bytes data
Sun Nov 10 20:08:50 2024 (466032302): Sending optional PCB data
Sun Nov 10 20:08:50 2024 (466032303): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:08:50 2024 (466032304): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:08:51 2024 (466033994): Heishamon stats: Uptime: 5 days 9 hours 27 minutes 13 seconds ## Free memory: 69% ## Free PSRAM: 2068860 bytes ## Free heap: 236160 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 2 ## Correct data: 4.31% Rules active: 0
Sun Nov 10 20:08:51 2024 (466033996): Requesting new panasonic data
Sun Nov 10 20:08:51 2024 (466033997): Already sending data. Buffering this send request
Sun Nov 10 20:08:51 2024 (466033998): Requesting new panasonic extra data
Sun Nov 10 20:08:51 2024 (466033998): Already sending data. Buffering this send request
Sun Nov 10 20:08:52 2024 (466034306): Previous read data attempt failed due to timeout!
Sun Nov 10 20:08:52 2024 (466034306): Received 0 bytes data
Sun Nov 10 20:08:52 2024 (466034307): Sending command from buffer
Sun Nov 10 20:08:52 2024 (466034308): sent bytes: 111 including checksum value: 18 
Sun Nov 10 20:08:52 2024 (466034309): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Nov 10 20:08:52 2024 (466034310): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Nov 10 20:08:52 2024 (466034310): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Nov 10 20:08:52 2024 (466034311): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
Sun Nov 10 20:12:09 2024 (466231332): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sun Nov 10 20:12:11 2024 (466233334): Previous read data attempt failed due to timeout!
Sun Nov 10 20:12:11 2024 (466233334): Received 0 bytes data
Sun Nov 10 20:12:11 2024 (466233335): Sending optional PCB data
Sun Nov 10 20:12:11 2024 (466233336): sent bytes: 20 including checksum value: 164 
Sun Nov 10 20:12:11 2024 (466233337): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
`

MiG-41, 2024-11-10:
Looks like another/new way of presenting this data. 
What model you have ?

alf1964m, 2024-11-10:
in the list of Heishamon 
TOP92 | Heat_Pump_Model | 38 | IDU:WH-SDC0509L3E5 ODU:WH-WDG09LE5
it is a Panasonic 9kW Series L

alf1964m, 2024-11-20:
is there anything news for this Solution - or no way to get the Head_Power_Consumption and Head_Power_Production valves?

alf1964m, 2024-11-20:
sorry for my questions:
in the xtop values list: 
XTOP0	Heat_Power_Consumption_Extra	-1	Watt
XTOP1	Cool_Power_Consumption_Extra	-1	Watt
XTOP2	DHW_Power_Consumption_Extra	-1	Watt
XTOP3	Heat_Power_Production_Extra	-1	Watt
XTOP4	Cool_Power_Production_Extra	-1	Watt
XTOP5	DHW_Power_Production_Extra	-1	Watt
this ist not better as -200 Watt in the top38, top39
may i can change anything?

geduxas, 2024-11-20:
> sorry for my questions: in the xtop values list: XTOP0 Heat_Power_Consumption_Extra -1 Watt XTOP1 Cool_Power_Consumption_Extra -1 Watt XTOP2 DHW_Power_Consumption_Extra -1 Watt XTOP3 [...]

Yes, because you're not receiving extra packet from pump (I don't see it in hex) but what I see is a lot of failed requests.. have you enabled pcb emulation? Do you have additional board? TAW adapter? Or do you constantly send values to heatpump?

alf1964m, 2024-11-20:
have you enabled pcb emulation? yes i to. 
and i got an TAW1 Adapter connect on the second connector Heishamon large patine.
I dont know any send order. can i check this anyway?

alf1964m, 2024-11-20:
only this too things are enable - 
Enable WiFi hotspot when not connected: 	
Debug log to MQTT topic from start: 	
Debug log hexdump enable from start: 	
Debug log USB: 	
Emulate optional PCB (does not work in listen only mode): X  <<<<<<
Enable opentherm processing: 	
Enable cztaw proxy port: X <<<<<<<
Listen only (parallel CZ-TAW1): 	
Use 1wire DS18b20: 	
Use s0 kWh metering:

geduxas, 2024-11-21:
You have pending commands in buffer which are not transmitted or failed receive acknowledge, it could be some bugs in heishamon or something we don't know right now. 
So try disable taw and optional pcb emulation, power cycle heatpump (turning completely off) and check again for extra topics, also share your findings and log with hex dump.

alf1964m, 2024-12-22:
Is there any news for a solution at this point - or no way to get the Head_Power_Consumption and Head_Power_Production values? I calculate the values outside the PCB. But it's not comfortable.

geduxas, 2024-12-22:
You haven't responded to latest message. Without your help it will nor be able to diagnose problem

IgorYbema, 2024-12-22:
I see no PROXY messages so it seems that the TAW1 isn't working either.
Try without optional pcb first to see if it works.
And if you can, please try this beta firmware: https://github.com/IgorYbema/HeishaMon/actions/runs/12358893896/artifacts/2327824158 
use the ESP32 file for the large heishamon

alf1964m, 2024-12-28:
Hello,
Please excuse my delay. first I disconnected the TAW1 adapter. There was no change to be seen.
Then I installed your beta firmware. HeishaMon/actions/runs/12358893896/artifacts/2327824158
Use the ESP32 file for the large heishamon
- Unfortunately, the values for
TOP15 Heat_Power_Production -200 and 
TOP16 Head_Power_Consumtion -200 still in the display. 
should I go back to the last firmware now?

Is there anything else I can do to correct this false display? Greetings and a Happy New Year

geduxas, 2024-12-28:
You should look for extended topic's from k series panasonic have moved those reading to other location. 
Also if you give latest log with hex dump it could help a lot

MiG-41, 2024-12-28:
@alf1964m ,yes please make now log , it can be only few complete answers from HP during work .

alf1964m, 2024-12-28:
i will do my verry best. 
Sat Dec 28 16:30:54 2024 (2680930): Heishamon stats: Uptime: 0 days 0 hours 44 minutes 40 seconds ## Free memory: 68% ## Free PSRAM: 2073488 bytes ## Free heap: 238344 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 16.13% Rules active: 0
Sat Dec 28 16:30:54 2024 (2680933): Requesting new panasonic data
Sat Dec 28 16:30:54 2024 (2680934): Already sending data. Buffering this send request
Sat Dec 28 16:30:54 2024 (2680935): Requesting new panasonic extra data
Sat Dec 28 16:30:54 2024 (2680936): Already sending data. Buffering this send request
Sat Dec 28 16:30:55 2024 (2682164): Previous read data attempt failed due to timeout!
Sat Dec 28 16:30:55 2024 (2682165): Received 0 bytes data
Sat Dec 28 16:30:55 2024 (2682166): PROXY Received 111 bytes
Sat Dec 28 16:30:55 2024 (2682167): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682167): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682168): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682169): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 
Sat Dec 28 16:30:55 2024 (2682170): PROXY Checksum and header received ok!
Sat Dec 28 16:30:55 2024 (2682171): PROXY requests basic data
Sat Dec 28 16:30:55 2024 (2682305): Sending command from buffer
Sat Dec 28 16:30:55 2024 (2682306): sent bytes: 111 including checksum value: 18 
Sat Dec 28 16:30:55 2024 (2682307): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682307): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682308): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:55 2024 (2682309): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:56 2024 (2682769): Received 203 bytes data
Sat Dec 28 16:30:56 2024 (2682770): data: 71 C8 01 10 56 55 52 61 00 55 00 15 00 00 00 00 00 00 00 00 6A 15 13 55 96 15 55 59 65 19 56 00 
Sat Dec 28 16:30:56 2024 (2682770): data: 00 00 00 00 00 00 9A 92 80 80 B2 71 71 BD 99 00 00 00 00 00 00 00 00 00 00 00 80 88 5B 8A 85 85 
Sat Dec 28 16:30:56 2024 (2682771): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AA 9B 76 8F B7 A3 7B 8F 94 85 78 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sat Dec 28 16:30:56 2024 (2682772): data: 8A 11 56 78 BC 06 80 80 1F 7E 7F 00 00 00 95 75 55 21 78 15 59 05 1B 13 65 21 51 15 01 3F 9B F2 
Sat Dec 28 16:30:56 2024 (2682773): data: 00 E2 D5 0B 34 99 83 92 0C 29 98 96 00 B4 80 9A 9D 32 32 A4 B2 32 32 32 9A A4 9F 9E 96 9B 73 00 
Sat Dec 28 16:30:56 2024 (2682774): data: 74 00 00 33 01 0F 1F 00 00 CD 09 24 53 3A 01 49 01 01 01 FF 04 00 23 05 00 01 00 00 01 00 00 0A 
Sat Dec 28 16:30:56 2024 (2682774): data: 01 00 00 00 00 00 00 04 80 17 43 
Sat Dec 28 16:30:56 2024 (2682775): Checksum and header received ok!
Sat Dec 28 16:30:56 2024 (2682776): received TOP1 Pump_Flow: 9.80
Sat Dec 28 16:30:56 2024 (2682779): received TOP6 Main_Outlet_Temp: 29.50
Sat Dec 28 16:30:56 2024 (2682782): received TOP53 Eva_Outlet_Temp: -12
Sat Dec 28 16:30:56 2024 (2682785): received TOP118 Second_Room_Thermostat_Temp: -128
Sat Dec 28 16:30:56 2024 (2682790): Sending command from buffer
Sat Dec 28 16:30:56 2024 (2682791): sent bytes: 111 including checksum value: 1 
Sat Dec 28 16:30:56 2024 (2682791): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:56 2024 (2682792): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:56 2024 (2682793): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:56 2024 (2682794): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:58 2024 (2684796): Previous read data attempt failed due to timeout!
Sat Dec 28 16:30:58 2024 (2684797): Received 0 bytes data
Sat Dec 28 16:30:58 2024 (2684797): Sending optional PCB data
Sat Dec 28 16:30:58 2024 (2684798): sent bytes: 20 including checksum value: 164 
Sat Dec 28 16:30:58 2024 (2684799): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sat Dec 28 16:30:58 2024 (2685040): PROXY Received 111 bytes
Sat Dec 28 16:30:58 2024 (2685041): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:58 2024 (2685042): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:58 2024 (2685042): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:30:58 2024 (2685043): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 
Sat Dec 28 16:30:58 2024 (2685044): PROXY Checksum and header received ok!
Sat Dec 28 16:30:58 2024 (2685045): PROXY requests extra data
Sat Dec 28 16:31:00 2024 (2686801): Previous read data attempt failed due to timeout!
Sat Dec 28 16:31:00 2024 (2686802): Received 0 bytes data
Sat Dec 28 16:31:00 2024 (2686802): Sending optional PCB data
Sat Dec 28 16:31:00 2024 (2686803): sent bytes: 20 including checksum value: 164 
[... cut ...]
Sat Dec 28 16:31:21 2024 (2708328): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:31:21 2024 (2708329): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:31:21 2024 (2708330): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 
Sat Dec 28 16:31:21 2024 (2708331): PROXY Checksum and header received ok!
Sat Dec 28 16:31:21 2024 (2708332): PROXY requests extra data

geduxas, 2024-12-28:
I still see proxy which is making it wors, and don't see any extra data, disable proxy and remove taw1 adapter

MiG-41, 2024-12-28:
And myybe also OptionalPCB  , if it is turn on emulation ,turn it off.

alf1964m, 2024-12-28:
this is disconnect the TAW1 Adapter again and reboot / new start 
Sat Dec 28 16:47:46 2024 (8460): Previous read data attempt failed due to timeout!
Sat Dec 28 16:47:46 2024 (8461): Received 0 bytes data
Sat Dec 28 16:47:46 2024 (8462): Sending optional PCB data
Sat Dec 28 16:47:46 2024 (8463): sent bytes: 20 including checksum value: 164 
Sat Dec 28 16:47:46 2024 (8464): data: F1 11 01 50 00 00 40 FF FF E5 FF FF 00 FF EB FF FF 00 00 
Sat Dec 28 16:47:48 2024 (10001): Connecting to mqtt server ...
Sat Dec 28 16:47:48 2024 (10633): Heishamon stats: Uptime: 0 days 0 hours 0 minutes 10 seconds ## Free memory: 76% ## Free PSRAM: 2088092 bytes ## Free heap: 265704 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 0.00% Rules active: 0
Sat Dec 28 16:47:48 2024 (10635): Requesting new panasonic data
Sat Dec 28 16:47:48 2024 (10636): Already sending data. Buffering this send request
Sat Dec 28 16:47:48 2024 (10647): Previous read data attempt failed due to timeout!
Sat Dec 28 16:47:48 2024 (10648): Received 0 bytes data
Sat Dec 28 16:47:48 2024 (10649): Sending command from buffer
Sat Dec 28 16:47:48 2024 (10650): sent bytes: 111 including checksum value: 18 
Sat Dec 28 16:47:48 2024 (10651): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:47:48 2024 (10652): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:47:48 2024 (10653): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:47:48 2024 (10653): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 16:47:49 2024 (11124): Received 203 bytes data
Sat Dec 28 16:47:49 2024 (11124): data: 71 C8 01 10 56 55 52 61 00 55 00 15 00 00 00 00 00 00 00 00 6A 15 13 55 96 15 55 59 65 19 56 00 
Sat Dec 28 16:47:49 2024 (11125): data: 00 00 00 00 00 00 9A 92 80 80 B2 71 71 BD 99 00 00 00 00 00 00 00 00 00 00 00 80 88 5B 8A 85 85 
Sat Dec 28 16:47:49 2024 (11126): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AA 9B 76 8F B7 A3 7B 8F 94 85 78 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sat Dec 28 16:47:49 2024 (11127): data: 8A 11 56 78 BC 06 80 80 1F 7E 7F 00 00 00 95 75 55 21 78 15 59 05 09 13 65 21 51 15 01 42 9A FD 
Sat Dec 28 16:47:49 2024 (11127): data: 00 E2 D5 0B 34 99 83 92 0C 29 98 96 00 B4 80 99 99 32 32 A4 B2 32 32 32 9A A4 9A 96 96 8E 7F 00 
Sat Dec 28 16:47:49 2024 (11128): data: 83 00 00 15 01 01 01 00 00 58 1B 5A A7 01 01 79 01 C9 C9 FF 04 00 23 05 00 01 00 00 01 00 00 0A 
Sat Dec 28 16:47:49 2024 (11129): data: 01 00 00 00 00 00 00 04 80 41 BE 
Sat Dec 28 16:47:49 2024 (11130): Checksum and header received ok!
Sat Dec 28 16:47:49 2024 (11131): received TOP0 Heatpump_State: 1
Sat Dec 28 16:47:49 2024 (11133): received TOP1 Pump_Flow: 27.34
Sat Dec 28 16:47:49 2024 (11136): received TOP2 Force_DHW_State: 0
Sat Dec 28 16:47:49 2024 (11139): received TOP3 Quiet_Mode_Schedule: 0
Sat Dec 28 16:47:49 2024 (11141): received TOP4 Operating_Mode_State: 0
Sat Dec 28 16:47:49 2024 (11144): received TOP5 Main_Inlet_Temp: 25
Sat Dec 28 16:47:49 2024 (11147): received TOP6 Main_Outlet_Temp: 25
Sat Dec 28 16:47:49 2024 (11149): received TOP7 Main_Target_Temp: 36
Sat Dec 28 16:47:49 2024 (11152): received TOP8 Compressor_Freq: 0
Sat Dec 28 16:47:49 2024 (11155): received TOP9 DHW_Target_Temp: 50
Sat Dec 28 16:47:49 2024 (11158): received TOP10 DHW_Temp: 52
Sat Dec 28 16:47:49 2024 (11160): received TOP11 Operations_Hours: 1314
Sat Dec 28 16:47:49 2024 (11163): received TOP12 Operations_Counter: 1278
Sat Dec 28 16:47:49 2024 (11166): received TOP13 Main_Schedule_State: 0
Sat Dec 28 16:47:49 2024 (11168): received TOP14 Outside_Temp: 0
Sat Dec 28 16:47:49 2024 (11171): received TOP15 Heat_Power_Production: -200
Sat Dec 28 16:47:49 2024 (11174): received TOP16 Heat_Power_Consumption: -200
Sat Dec 28 16:47:49 2024 (11176): received TOP17 Powerful_Mode_Time: 0
Sat Dec 28 16:47:49 2024 (11178): received TOP18 Quiet_Mode_Level: 3
Sat Dec 28 16:47:49 2024 (11179): received TOP19 Holiday_Mode_State: 0
Sat Dec 28 16:47:49 2024 (11181): received TOP20 ThreeWay_Valve_State: 0
Sat Dec 28 16:47:49 2024 (11182): received TOP21 Outside_Pipe_Temp: -1
Sat Dec 28 16:47:49 2024 (11184): received TOP22 DHW_Heat_Delta: -8
Sat Dec 28 16:47:49 2024 (11185): received TOP23 Heat_Delta: 5
Sat Dec 28 16:47:49 2024 (11187): received TOP24 Cool_Delta: 5
Sat Dec 28 16:47:49 2024 (11188): received TOP25 DHW_Holiday_Shift_Temp: -15
Sat Dec 28 16:47:49 2024 (11190): received TOP26 Defrosting_State: 0
Sat Dec 28 16:47:49 2024 (11191): received TOP27 Z1_Heat_Request_Temp: 26
Sat Dec 28 16:47:49 2024 (11193): received TOP28 Z1_Cool_Request_Temp: 18
Sat Dec 28 16:47:49 2024 (11194): received TOP29 Z1_Heat_Curve_Target_High_Temp: 42
Sat Dec 28 16:47:49 2024 (11196): received TOP30 Z1_Heat_Curve_Target_Low_Temp: 27
Sat Dec 28 16:47:49 2024 (11197): received TOP31 Z1_Heat_Curve_Outside_High_Temp: 15
Sat Dec 28 16:47:49 2024 (11199): received TOP32 Z1_Heat_Curve_Outside_Low_Temp: -10
Sat Dec 28 16:47:49 2024 (11200): received TOP33 Room_Thermostat_Temp: 22
Sat Dec 28 16:47:49 2024 (11202): received TOP34 Z2_Heat_Request_Temp: 0
Sat Dec 28 16:47:49 2024 (11203): received TOP35 Z2_Cool_Request_Temp: 0
Sat Dec 28 16:47:49 2024 (11205): received TOP36 Z1_Water_Temp: -78
Sat Dec 28 16:47:49 2024 (11206): received TOP37 Z2_Water_Temp: -78
[... cut ...]
Sat Dec 28 16:48:08 2024 (30005): Heishamon stats: Uptime: 0 days 0 hours 0 minutes 30 seconds ## Free memory: 69% ## Free PSRAM: 2074732 bytes ## Free heap: 241756 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 13.33% Rules active: 0
Sat Dec 28 16:48:08 2024 (30007): Requesting new panasonic data
Sat Dec 28 16:48:08 2024 (30008): Already sending data. Buffering this send request
Sat Dec 28 16:48:08 2024 (30009): Requesting new panasonic extra data
Sat Dec 28 16:48:08 2024 (30009): Already sending data. Buffering this send request

geduxas, 2024-12-28:
As @MiG-41 told, also turn off emulation, also please imclude hex log after "Requesting new panasonic extra data" after this only it will get additional data where could be your power values.. also if you could, take photos of heishamon connection in heatpump.

IgorYbema, 2024-12-28:
I do see the 'extra data request' being sent but it ends with a timeout. I am guessing that the connection to the heatpump is on the wrong port. Please send photo's indeed and disable emulation for optional pcb.

alf1964m, 2024-12-28:
[image]

alf1964m, 2024-12-28:
[image]

IgorYbema, 2024-12-28:
Ok cable seems to be in the correct position (CN-NMODE on the optional pcb). As you have the optional PCB there is no reason to enable the 'emulation' also which you had enabled. This could have confused the optional pcb you have installed. Please try again with the emulation now disabled. Power off and power on the heatpump first to reset the heatpump

alf1964m, 2024-12-28:
[image]

alf1964m, 2024-12-28:
[image]
[image]
is it possible that the cables are swapped?

geduxas, 2024-12-28:
@alf1964m plase do as @IgorYbema told, disable emulation, powercycle whole pump, and post your logs.

alf1964m, 2024-12-28:
Heat pump off for 5 minutes. fuses also off. 
Restart.

Sat Dec 28 17:32:53 2024 (320107): Heishamon stats: Uptime: 0 days 0 hours 5 minutes 20 seconds ## Free memory: 68% ## Free PSRAM: 2073948 bytes ## Free heap: 238724 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 94.83% Rules active: 0
Sat Dec 28 17:32:53 2024 (320110): Requesting new panasonic data
Sat Dec 28 17:32:53 2024 (320111): sent bytes: 111 including checksum value: 18 
Sat Dec 28 17:32:53 2024 (320111): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:53 2024 (320112): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:53 2024 (320113): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:53 2024 (320114): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:53 2024 (320115): Requesting new panasonic extra data
Sat Dec 28 17:32:53 2024 (320115): Already sending data. Buffering this send request
Sat Dec 28 17:32:54 2024 (320510): Received 203 bytes data
Sat Dec 28 17:32:54 2024 (320511): data: 71 C8 01 10 56 55 52 61 00 55 00 15 00 00 00 00 00 00 00 00 6A 15 13 55 96 15 55 59 65 19 56 00 
Sat Dec 28 17:32:54 2024 (320511): data: 00 00 00 00 00 00 9A 92 80 80 B2 71 71 BD 99 00 00 00 00 00 00 00 00 00 00 00 80 88 5B 8A 85 85 
Sat Dec 28 17:32:54 2024 (320512): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AA 9B 76 8F B7 A3 7B 8F 94 85 78 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sat Dec 28 17:32:54 2024 (320513): data: 8A 11 56 78 BC 06 80 80 1F 7E 7F 00 00 00 95 75 55 21 78 15 59 05 1B 13 65 21 51 15 01 3F 9A F3 
Sat Dec 28 17:32:54 2024 (320514): data: 00 E2 D5 0B 34 99 83 92 0C 29 98 95 00 B3 80 99 9B 32 32 A4 B2 32 32 32 9A A4 9D 9C 95 9A 7A 00 
Sat Dec 28 17:32:54 2024 (320515): data: 7E 00 00 32 01 10 1F 00 00 DF 0A 27 57 3A 01 54 01 B9 B9 00 05 00 23 05 00 01 00 00 01 00 00 0A 
Sat Dec 28 17:32:54 2024 (320515): data: 01 00 00 00 00 00 00 04 80 17 A6 
Sat Dec 28 17:32:54 2024 (320516): Checksum and header received ok!
Sat Dec 28 17:32:54 2024 (320517): received TOP6 Main_Outlet_Temp: 27.50
Sat Dec 28 17:32:54 2024 (320520): received TOP10 DHW_Temp: 51
Sat Dec 28 17:32:54 2024 (320522): received TOP21 Outside_Pipe_Temp: -6
Sat Dec 28 17:32:54 2024 (320525): received TOP50 Discharge_Temp: 28
Sat Dec 28 17:32:54 2024 (320529): received TOP93 Pump_Duty: 86
Sat Dec 28 17:32:54 2024 (320533): Sending command from buffer
Sat Dec 28 17:32:54 2024 (320534): sent bytes: 111 including checksum value: 1 
Sat Dec 28 17:32:54 2024 (320535): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320536): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320537): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320538): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320996): Received 203 bytes data
Sat Dec 28 17:32:54 2024 (320997): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 AD 02 01 00 01 00 53 06 01 00 01 00 01 00 00 00 01 00 
Sat Dec 28 17:32:54 2024 (320997): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320998): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (320999): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (321000): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (321001): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:32:54 2024 (321001): data: 00 00 00 00 00 00 00 00 00 00 1D 
Sat Dec 28 17:32:54 2024 (321002): Checksum and header received ok!
Sat Dec 28 17:32:54 2024 (321003): received XTOP0 Heat_Power_Consumption_Extra: 684
Sat Dec 28 17:32:54 2024 (321006): received XTOP3 Heat_Power_Production_Extra: 1618
Sat Dec 28 17:33:03 2024 (330109): Heishamon stats: Uptime: 0 days 0 hours 5 minutes 30 seconds ## Free memory: 69% ## Free PSRAM: 2074232 bytes ## Free heap: 241604 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 95.00% Rules active: 0
Sat Dec 28 17:33:03 2024 (330111): Requesting new panasonic data
Sat Dec 28 17:33:03 2024 (330112): sent bytes: 111 including checksum value: 18 
Sat Dec 28 17:33:03 2024 (330113): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:03 2024 (330114): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:03 2024 (330114): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:03 2024 (330115): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:03 2024 (330116): Requesting new panasonic extra data
Sat Dec 28 17:33:03 2024 (330117): Already sending data. Buffering this send request
[... cut ...]
Sat Dec 28 17:33:14 2024 (341066): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:14 2024 (341067): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 17:33:14 2024 (341067): data: 00 00 00 00 00 00 00 00 00 00 4C 
Sat Dec 28 17:33:14 2024 (341068): Checksum and header received ok!
Sat Dec 28 17:33:14 2024 (341069): received XTOP0 Heat_Power_Consumption_Extra: 682
Sat Dec 28 17:33:14 2024 (341072): received XTOP3 Heat_Power_Production_Extra: 1828

alf1964m, 2024-12-28:
<html><body>

XTOP0 | Heat_Power_Consumption_Extra | 690 | Watt
-- | -- | -- | --
XTOP1 | Cool_Power_Consumption_Extra | 0 | Watt
XTOP2 | DHW_Power_Consumption_Extra | 0 | Watt
XTOP3 | Heat_Power_Production_Extra | 2886 | Watt
XTOP4 | Cool_Power_Production_Extra | 0 | Watt

geduxas, 2024-12-28:
Yes, you have it. So now need to find out where probem was.. i suspect same as @IgorYbema you have optional pcb installed also have emulation enabled.. so it could make some problems.. so try to avoid emulation, and try to bring back your setup while monitoring these xtop values.. if it's disapperes in some step it could be source of problem's

alf1964m, 2024-12-28:
Thank you. Should I now also connect the TAW1 adapter again and leave the test firmware on it?

MiG-41, 2024-12-28:
Test firmware is ok to leave.

Exacly where you are going to connect CZ-TAW1 plug ?

alf1964m, 2024-12-28:
I had plugged in the TAW1 adapter according to the instructions on the Heishamon board

geduxas, 2024-12-28:
> Test firmware is ok to leave.
> 
> Exacly where you are going to connect CZ-TAW1 plug ?

Big board can work as proxy for taw1 so it should be connected there, also enabled proxy

alf1964m, 2024-12-28:
without the test firmware, the XTOP0 does not have an "Extra" extension. Then no consumption is displayed again? 
Could you please write me more precisely where I should turn on a proxy? (on the setting page of the Heishamon?)
then I plug the cable of the TAW1 adapter directly onto the expansion board in the heat pump?
but I have to buy a suitable cable first.

MiG-41, 2024-12-28:
https://github.com/Egyras/HeishaMon/issues/578#issuecomment-2564368346

You show this settings , there is "enable cztaw proxy port"

alf1964m, 2024-12-28:
i need every time a few seconds to translate :-) 
I activate the proxy on the settings page. 
then I switch off the heat pump and turn off the fuses. Plug the connection cable of the TAW1 adapter back into the slot of the Heishamon board and restart the heat pump. 
I report whether it still shows the consumption.

alf1964m, 2024-12-28:
You are the best specialists. Thanks. it now also seems that the missing consumption and generation values go to the XTOP0 with the "Extra" extension. 
[image]

alf1964m, 2024-12-28:
for your info
________________________
Sat Dec 28 18:33:20 2024 (621151): Received 203 bytes data
Sat Dec 28 18:33:20 2024 (621152): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 CD 03 01 00 01 00 06 10 01 00 01 00 01 00 00 00 01 00 
Sat Dec 28 18:33:20 2024 (621153): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621154): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621155): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621155): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621156): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621157): data: 00 00 00 00 00 00 00 00 00 00 3F 
Sat Dec 28 18:33:20 2024 (621158): Checksum and header received ok!
Sat Dec 28 18:33:20 2024 (621158): received XTOP0 Heat_Power_Consumption_Extra: 972
Sat Dec 28 18:33:20 2024 (621161): received XTOP3 Heat_Power_Production_Extra: 4101
Sat Dec 28 18:33:20 2024 (621419): PROXY Received 111 bytes
Sat Dec 28 18:33:20 2024 (621419): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621420): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621421): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:20 2024 (621422): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 
Sat Dec 28 18:33:20 2024 (621422): PROXY Checksum and header received ok!
Sat Dec 28 18:33:20 2024 (621423): PROXY requests extra data
Sat Dec 28 18:33:23 2024 (624317): PROXY Received 111 bytes
Sat Dec 28 18:33:23 2024 (624318): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:23 2024 (624319): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:23 2024 (624320): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:23 2024 (624320): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 
Sat Dec 28 18:33:23 2024 (624321): PROXY Checksum and header received ok!
Sat Dec 28 18:33:23 2024 (624322): PROXY requests basic data
Sat Dec 28 18:33:26 2024 (627222): PROXY Received 111 bytes
Sat Dec 28 18:33:26 2024 (627223): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:26 2024 (627224): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:26 2024 (627225): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:26 2024 (627225): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 
Sat Dec 28 18:33:26 2024 (627226): PROXY Checksum and header received ok!
Sat Dec 28 18:33:26 2024 (627227): PROXY requests extra data
Sat Dec 28 18:33:29 2024 (630134): PROXY Received 111 bytes
Sat Dec 28 18:33:29 2024 (630135): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630136): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630137): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630137): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 
Sat Dec 28 18:33:29 2024 (630138): PROXY Checksum and header received ok!
Sat Dec 28 18:33:29 2024 (630139): PROXY requests basic data
Sat Dec 28 18:33:29 2024 (630274): Heishamon stats: Uptime: 0 days 0 hours 10 minutes 30 seconds ## Free memory: 68% ## Free PSRAM: 2074444 bytes ## Free heap: 240056 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (10.63.173.45) ## Mqtt reconnects: 1 ## Correct data: 100.00% Rules active: 0
Sat Dec 28 18:33:29 2024 (630277): Requesting new panasonic data
Sat Dec 28 18:33:29 2024 (630278): sent bytes: 111 including checksum value: 18 
Sat Dec 28 18:33:29 2024 (630279): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630280): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630281): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630281): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:33:29 2024 (630282): Requesting new panasonic extra data
Sat Dec 28 18:33:29 2024 (630283): Already sending data. Buffering this send request
Sat Dec 28 18:33:29 2024 (630736): Received 203 bytes data
Sat Dec 28 18:33:29 2024 (630737): data: 71 C8 01 10 56 55 52 59 00 55 00 15 00 00 00 00 00 00 00 00 6A 15 13 55 96 15 55 59 65 19 56 00 
Sat Dec 28 18:33:29 2024 (630738): data: 00 00 00 00 00 00 9C 92 80 80 B2 71 71 BD 99 00 00 00 00 00 00 00 00 00 00 00 80 88 5B 8A 85 85 
[... cut ...]
Sat Dec 28 18:34:21 2024 (682449): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:34:21 2024 (682450): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:34:21 2024 (682451): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Dec 28 18:34:21 2024 (682452): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 
Sat Dec 28 18:34:21 2024 (682452): PROXY Checksum and header received ok!
Sat Dec 28 18:34:21 2024 (682453): PROXY requests basic data

MiG-41, 2024-12-28:
So finally , what was wrong ?

geduxas, 2024-12-28:
It's still working, so I thin optional pcb emulation was problem..

alf1964m, 2024-12-28:
That was just for information
I think that nothing is wrong anymore, until the next update of the firmware.

closed 2025-02-02 by github-actions[bot]