heishamon/HeishaMon #579: How to add relays in v0.5?

kamil2466, 2024-11-11:
Do you guys know how to add relays so they can be controlled in Home Assistant? I need to add 1 relay to control the circulating pump. I've looked on the forums but don't see anywhere how to add it.

MiG-41, 2024-11-11:
Quote from ReadMe:

`To control the relay just send a value of 1 or 0 to the MQTT topic "panasonic_heat_pump/gpio/relay/one" for relay one or "panasonic_heat_pump/gpio/relay/two" for relay two.`

You have to made siutable to this entities.

kamil2466, 2024-11-11:
Can you send an example of how to do this for Home Assistant, I see 2 not available entities for relays in HA

McMagellan, 2024-11-11:
And how can you control the relays from Rules and also query the status?

Is it possible to control the relays like a Shelly relay with: 
`http://192.168.xxx.xxx/relay/0?turn=on `

as a simple local URL command like: 
`http://192.168.178.36/command? SetZ1HeatRequestTemperature=1`

closed 2024-12-19 by github-actions[bot]