heishamon/HeishaMon #582: Heishamon V5: Debug connection triggers reset

McMagellan, 2024-11-13:
For a few days now I have been able to record debug information via the USB port and it works as expected.
I can disconnect the existing connection by removing the USB plug on the PC without any consequences for Heishamon.
I use PUTTY as a terminal program with the ability to record the data in a log file.
If I close the PUTTY terminal window, a reset is triggered in the Heishamon, the uptime restarts and the ruleset is not started again. 
It looks like a cancel signal is being sent by PUTTY and Heishamon cannot process it.
I know this is a special thing that doesn't normally occur in normal operation. I would still like to point this out.

CurlyMoo, 2024-11-16:
This behavior is what i see in all my ESP's. So it doesn't seem to be anything special to the HeishaMon.

McMagellan, 2024-11-16:
The special thing about Heishamon is that after such a reset by terminating the terminal program (Putty) at the debug port, the ruleset no longer starts. If it were just a matter of restarting the uptime and also restarting the rules, you could live with it.

I log the debug connection permanently and so that the file doesn't get too large, I create a new file every 2 days or as needed. The way I do it now is that I first pull out the USB plug on the laptop, which causes the connection to be interrupted. This is how it works without any problems.

Another question: 
What about the relays on Heishamon5? Can you give a small example of how to switch these from iobroker.

closed 2024-12-17 by McMagellan