heishamon/HeishaMon #584: Protocol description for SG is mixup up

Celphor, 2024-11-17:
Desciptions in the decyption document for byte 71 and 72 are swapped, as well as 73 and 74 

Correct is:

71 = SG Setting 1 DHW Capacity
72 = SG Setting 1 Heating Capacity
73 = SG Setting 2 DHW Capacity
74 = SG Setting 2 Heating Capacity

I can confirm SG works according to the manual when external SG switches are applied

geduxas, 2024-11-17:
It's only in documentation or in code also?

Celphor, 2024-11-17:
Only in documentation. There is no TOP for it.

There seem to be no shadow bits für the SG inputs in the protocol. At least i didn't find any changes in the heatpump reply.

Also the SG function is pretty useless for heishamon users. Increasing/decreasing heatcurve or dhw temp shifts can be derived with existing TOPs much better. No need to put any effort on it

closed 2024-12-24 by github-actions[bot]