heishamon/HeishaMon #585: Can you please help diagnose my pump?

HighlyCompressedAir, 2024-11-17:
During heating the discharge temperature of my pump is much higher than expected and fluctuates. My friend has a TCAP H and his discharge temperature is rock solid, usually about 6-8 degrees higher than outlet water temperature. In my case, HP K model, discharge starts at 6 degrees higher than outlet and then rises to 25 and even 50 higher than outlet in some cases, reaching over 90 degrees Celsius.
Can you please check how it looks like with your pumps? I wonder if something is wrong with mine.

Yesterday it took almost 3 hours to heat up the tank instead of the usual 1 hour. In the middle of heating, COP fell down from 4 to 1 and even 0,8 and stayed like that for about 90 minutes (heat production between 1 and 2 kw, compressor at 33 hz). Then it slowly recovered, tank heating finished and after switching to room heating, everything was fine. I hope that was just some kind of a glitch and it won't happen again, but it got me thinking about other things that may be wrong.

Other than that the pump appears to work fine, COP is pretty much on point with the specs. I would appreciate your help. I know it's not the right place for such questions, if you know some forums where I can ask please share.

geduxas, 2024-11-21:
As for tank reheat, do you use some automations for it? Maybe you left pump in quiet mode? Or some SG limits? Maybe demand control?

MiG-41, 2024-11-22:
Or maybe one of valves temporarly stuck...

geduxas, 2024-11-22:
I had, with demand control lowered COP dramatically while reheating DHW.

MiG-41, 2024-11-22:
Bur not below 1 .... That means that you are not taking any energy from outside , anly release it to the air.... It should not happen at normal work even at -20deg outside ,and outlet 55deg.

HighlyCompressedAir, 2024-11-22:
> Or maybe one of valves temporarly stuck... Quiet mode , SG ,demand will not couse COP values less then 1 ( not on long notice , since ofcourse during start it can happen).

That was my first thought, some valve doesn't operate properly. That's why discharge temp is so unstable. When that happened discharge was below outlet temperature. If you know anyone with HP K series that also uses grafana or service cloud and can post his graphs that would help a lot.
Since then the pump works fine, although I don't like the noises it makes. I don't know if it should work like that:
https://www.dropbox.com/scl/fi/6d3g36bi17jixwjfy784h/hpnoise.m4a?rlkey=fzuvbum2hje4hx4q45q72thfj&e=1&st=qkuzrfdk&dl=0

Does it sound normal to you?

Zaph9001, 2024-11-24:
I do have a K series HP and see the same behaviour for the discharge temperature and I was wondering myself why it is that unstable. But as your pump heating is OK, I did not see such strange behaviour for DHW heating though.
So, if you find out why discharge temp shows this, let us know here.
Noise wise it does not souns lije yours, I have rather a gas flow hissing noise.
[image]

HighlyCompressedAir, 2024-11-24:
Thank you, it does look a lot like my pump, which is good news I guess. It's unlikely that we both have the same issue, so maybe it is normal behavior. Maybe it's more efficient and that's why pump's algorithm does this? Who knows.
My DHW situatuation only happened this one time. I have monitoring data since September and I've seen only one other glitch that happened a couple months back also during DHW heating - compressor shutdown for like 30 seconds and then started operating again. Other than that the pump works fine.

Can you check your eva outlet temperature? On TCAP H it's always about 2 degrees higher than outdoor pipe temperature. On my pump it can be 5 degrees higher and sometimes 10 degrees lower, again no stability whatsoever.

Zaph9001, 2024-11-25:
After I rechecked, I found the fluctuation of the discharge temperature when the compressor frequency is changing. When everything is in stable condition, also the discharge temp is.
Difference in outside pipe temp to eva outlet temp is related to outdoor temperature and fan speed (silent mode), to my understanding. It should be stable during normal running but changes when pump switches mode or shuts off. Then the compressed gas ist released into the systems to protect the compressor.
But I am not an expert, this is just my understanding.
[image]
[image]

HighlyCompressedAir, 2024-11-25:
> After I rechecked, I found the fluctuation of the discharge temperature when the compressor frequency is changing. When everything is in stable condition, also the discharge temp is.

Not really. I marked a few periods of stable compressor frequency on your previous graph and as you can see discharge varies massively.
[image]

On your last chart that's also the case when discharge goes from 80 to 90 between 6:40 and 6:50 and then goes back to 80 at 7:00. It's just that 3h time window is too short, your previous chart was from 24h so it's more visible. Sometimes my discharge temperatures are also quite stable, within a few degrees, usually when it's warmer outside.

> Difference in outside pipe temp to eva outlet temp is related to outdoor temperature and fan speed (silent mode), to my understanding. It should be stable during normal running but changes when pump [...]
But I am not an expert, this is just my understanding.

Eva temp is measuring the refrigerant temp after it's been heated up by the fan, so it should always be higher than outdoor pipe, which measures the refrigerant before that happens. Eva temp falls below outdoor pipe when the heat exchanger is frozen, which I guess makes sense, as the fan can't heat it up too much. Soon after that happens outdoor pipe temp falls dramatically. Try to find a period when defrosts are happening and see if that happens with your pump too.

I found out that TCAP models work a little bit different than HP models as they can steer where the refrigerant goes so that some of it doesn't have to go through the entire cycle. That's why on TCAP refrigerant temperatures look smoother and there are no big variations. Downside is that it makes it harder to spot issues like refrigerant leak, because the pump tries to work with what it has. HP models are simply more primitive.

closed 2025-01-01 by github-actions[bot]