heishamon/HeishaMon #588: No output of local variables in the console

Binifada1956, 2024-11-23:
Earlier versions also took the local variables into account.
Is it possible to write both types of variables, i.e. local and global, to the console?

[image]

Binifada1956, 2024-12-24:
Problem is not solved 🤷‍♂️

IgorYbema, 2025-01-13:
which version are you using? I believe this was fixed in v3.8 already

Binifada1956, 2025-01-13:
> which version are you using? I believe this was fixed in v3.8 already

… 3.8 on ESP32 🤷‍♂️

IgorYbema, 2025-01-14:
It works fine. But understand that it only prints local variables seen in the system boot
```
on System#Boot then
   $localtest = 1;
   #globaltest = 1;
end
```

[image]

Binifada1956, 2025-01-14:
> It works fine. But understand that it only prints local variables seen in the system boot
>
> ```
> on System#Boot then
>    $localtest = 1;
>    #globaltest = 1;
> end
> ``` [...]

I understand that local variables should also be declared in the system boot. Only then will the content be displayed in the console.

closed 2025-01-14 by Binifada1956