heishamon/HeishaMon #589: WiFi Issues on Heishamon v5 (large board)

geduxas, 2024-11-23:
I have some wifi issues on new board, at home i have multiple AP with same or different names, after fresh boot heishamon finds only one AP. Some times it detects other wifi AP.
Problem is that i have multiplie AP with same name, and it won't detect nearest AP on reboot. So it will connect to AP with lowest RSSI, and will not reconect to anything else. 

Didn't had any issues with small version.

IgorYbema, 2024-11-23:
It should not depend on the BSSID of the wifi station, only the (named) SSID. It only stores this (plus password) so during runtime it should connect to a reasonable working wifi station with same SSID name. There is no smart logic to switch to a better RSSI if there is one later as long at can keep a working wifi with the first connected one.

At boot the large heishamon does a bit different scan of the wifi networks. It will only rescan new wifi if you are in the settings page. But after 10-30 seconds you should see wifi networks which it didn't see before.

IgorYbema, 2024-11-23:
I just read that the ESP32 connects to the first AP it finds with same SSID and doesn't look further on other channels. Is it true that your other stations are on other channels?

geduxas, 2024-11-23:
Yes AP is in different channels. But somehow it first funds that farest away AP. 

Also please take look in slack, i have additional hardware issues :)

geduxas, 2024-12-04:
So for a record, 
Heishamon Large board have some issues with wifi,
It will connect on first wifi network by it's name, so if in home are couple of AP with same name it will connect to first it finds (it also could be with lowest signal) need to clarify it could sort AP by channels. 
solution for that use different AP or if possible block heishamon in worst AP.

geduxas, 2024-12-04:
Thank's @IgorYbema , i checked current version, i am bit confused that scan all range will help, but maybe i will tray as soon as i can. here some forum i have found: 

https://www.esp32.com/viewtopic.php?f=19&t=18979

maybe WifiMulti library would make it better?

IgorYbema, 2024-12-07:
So if you have multi AP's with same SSID a full scan ofcourse helps instead of connecting to the first one it notices. Even if they are on the same channel but also on different channels. The previous code just connects to the first one it notices. This beta code lists all and then, should, connect to the strongest.

However it cannot do this while running. It only does this once during boot. Because a ESP doesn't have multiple PHY's so it can not scan and work at the same time. So while the heishamon board is connect to a wifi network it will never try to connect to a stronger one as long it stays connected to the weakest one.

Same for wifmulti library

geduxas, 2024-12-08:
I have tested, and yes this resolves issue.

closed 2025-01-13 by github-actions[bot]