heishamon/HeishaMon #591: Heishamon Homeassistant Commands are not working (Command is sent to MQRR)

julianher17, 2024-11-25:
Hi, 

I have a Problem with one of my 2 heatpumpts (both are running with heishamon).
If i sent SetForce HDW = 1 or SetOperationMode = 3 from Homeasssitant. These commands will not work in 90% of the cases.

I checked it in MQTT Explorer and I see that the commands are sent, but nothing happens.

Any idea what the problem could be?

MiG-41, 2024-11-25:
And what is the Correct received data value ? 
Perhaps there is a problem with communication , and most of data sent by Heishamon not comming back....
Can you look into log if on each Send command there is also Receive ?

And hope you are sending correct topics ( not SetForce HDW , but SetForceDHW ....)

julianher17, 2024-11-25:
Hi, 

- Data is also not displayed correctly in HA. For example: Right now Three valve state is in HA set to Tank. If I open the IP of the heishamon directly i see that the state is set to room.

- Topics are sent correct - sorry for my typo.

- If i send a command in HA i see it immediately in MQQT Explorer: commands.

- In MQTT Explorer I see: LWT = Offline

Sorry, but where can i check if there is a receive to my send command? As far as I can see in the MQTT Explorer is the command logged, but the states will not change. Really strange, because the interface is not offline I get all the data...

[image]

julianher17, 2024-11-25:
`Wifi signal: 54%
Memory free: 69%
Correct received data: 100%
MQTT reconnects: 3446
Uptime: 1 day 4 hours 43 minutes 11 seconds`

And in MQTT:

`stats = {"uptime":103435798,"voltage":3.41,"free memory":69,"free heap":26456,"wifi":52,"mqtt reconnects":3448,"total reads":41362,"good reads":41362,"bad crc reads":0,"bad header reads":0,"too short reads":0,"too long reads":0,"timeout reads":0,"version":"3.8","rules active":0}
`

MiG-41, 2024-11-25:
MQTT  reconnects 3446 , in 28h uptime gives averege time of reconnect every ~25s...  
Please ensure much better wifi signal , since now heishamon is focusing on recconection instead of doing his job...

julianher17, 2024-11-30:
Hi, 

I further investigated there - i have no 100% wifi, but lots of mqtt reconnects. 
I am not sure whats the probleme here, i am using the Homeassistant MQTT server.

As far as i can say this problem occured after adding a second heishamon to my network. Is there any problem with 2 heishamons?

In the console i see also every second:

`Already sending data. Buffering this send request`

geduxas, 2024-11-30:
You have 2 heishamon on single pump or couple? Also i have noticed that mqtt reconnect will be if heishamon fails receiving data from pump

julianher17, 2024-11-30:
Hey,

i have now 2 heishamon on 2 heatpumps in the same network.
Before i added the second heatpumpts there have been no problems.

for second heatpump i changed the mqtt topic base so thats not the same.

thanks a lot for your help!

julianher17, 2024-12-05:
So i got a new repeater and both heishamons have 100% wifi.

But i get a mqtt reconnect every 10 seconds. What could be the problem here?

geduxas, 2024-12-05:
Please share log's during mqtt reconect.

julianher17, 2024-12-05:
Sure:

`Thu Dec  5 14:07:38 2024 (6501348): Lost MQTT connection!
Thu Dec  5 14:07:38 2024 (6501350): Heishamon stats: Uptime: 0 days 1 hour 48 minutes 21 seconds ## Free memory: 76% ## Heap fragmentation: 20% ## Max free block: 23088 bytes ## Free heap: 29288 bytes ## Wifi: 100% (RSSI: -50) ## Mqtt reconnects: 217 ## Correct data: 100.00% Rules active: 0
Thu Dec  5 14:07:38 2024 (6501369): Requesting new panasonic data
Thu Dec  5 14:07:38 2024 (6501373): sent bytes: 111 including checksum value: 18 
Thu Dec  5 14:07:38 2024 (6501381): Requesting new panasonic extra data
Thu Dec  5 14:07:38 2024 (6501387): Already sending data. Buffering this send request
Thu Dec  5 14:07:39 2024 (6501805): Received 203 bytes data
Thu Dec  5 14:07:39 2024 (6501806): Checksum and header received ok!
Thu Dec  5 14:07:39 2024 (6501807): received TOP1 Pump_Flow: 15.89
Thu Dec  5 14:07:39 2024 (6501812): received TOP8 Compressor_Freq: 20
Thu Dec  5 14:07:39 2024 (6501821): received TOP115 Water_Pressure: 1.84
Thu Dec  5 14:07:39 2024 (6501825): received TOP118 Second_Room_Thermostat_Temp: -128
Thu Dec  5 14:07:39 2024 (6501832): Sending command from buffer
Thu Dec  5 14:07:39 2024 (6501838): sent bytes: 111 including checksum value: 1 
Thu Dec  5 14:07:39 2024 (6502305): Received 203 bytes data
Thu Dec  5 14:07:39 2024 (6502305): Checksum and header received ok!
Thu Dec  5 14:07:43 2024 (6506349): Lost MQTT connection!
Thu Dec  5 14:07:43 2024 (6506351): Heishamon stats: Uptime: 0 days 1 hour 48 minutes 26 seconds ## Free memory: 76% ## Heap fragmentation: 20% ## Max free block: 23088 bytes ## Free heap: 29248 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 217 ## Correct data: 100.00% Rules active: 0
Thu Dec  5 14:07:43 2024 (6506371): Requesting new panasonic data
Thu Dec  5 14:07:43 2024 (6506375): sent bytes: 111 including checksum value: 18 
Thu Dec  5 14:07:43 2024 (6506382): Requesting new panasonic extra data
Thu Dec  5 14:07:43 2024 (6506388): Already sending data. Buffering this send request
Thu Dec  5 14:07:44 2024 (6506805): Received 203 bytes data
Thu Dec  5 14:07:44 2024 (6506806): Checksum and header received ok!
Thu Dec  5 14:07:44 2024 (6506810): received TOP117 Economizer_Outlet_Temp: 107
Thu Dec  5 14:07:44 2024 (6506813): received TOP118 Second_Room_Thermostat_Temp: 127
Thu Dec  5 14:07:44 2024 (6506821): Sending command from buffer
Thu Dec  5 14:07:44 2024 (6506826): sent bytes: 111 including checksum value: 1 
Thu Dec  5 14:07:44 2024 (6507204): Received 203 bytes data
Thu Dec  5 14:07:44 2024 (6507205): Checksum and header received ok!
Thu Dec  5 14:07:48 2024 (6511350): Lost MQTT connection!
Thu Dec  5 14:07:48 2024 (6511352): Heishamon stats: Uptime: 0 days 1 hour 48 minutes 31 seconds ## Free memory: 76% ## Heap fragmentation: 20% ## Max free block: 23088 bytes ## Free heap: 29272 bytes ## Wifi: 100% (RSSI: -50) ## Mqtt reconnects: 217 ## Correct data: 100.00% Rules active: 0
Thu Dec  5 14:07:48 2024 (6511373): Requesting new panasonic data
Thu Dec  5 14:07:48 2024 (6511376): sent bytes: 111 including checksum value: 18 
Thu Dec  5 14:07:48 2024 (6511383): Requesting new panasonic extra data
Thu Dec  5 14:07:48 2024 (6511389): Already sending data. Buffering this send request
Thu Dec  5 14:07:49 2024 (6511805): Received 203 bytes data
Thu Dec  5 14:07:49 2024 (6511805): Checksum and header received ok!
Thu Dec  5 14:07:49 2024 (6511807): received TOP1 Pump_Flow: 15.53
Thu Dec  5 14:07:49 2024 (6511812): received TOP8 Compressor_Freq: 19
Thu Dec  5 14:07:49 2024 (6511821): received TOP115 Water_Pressure: 1.86
Thu Dec  5 14:07:49 2024 (6511824): received TOP118 Second_Room_Thermostat_Temp: -128
Thu Dec  5 14:07:49 2024 (6511832): Sending command from buffer
Thu Dec  5 14:07:49 2024 (6511837): sent bytes: 111 including checksum value: 1 
Thu Dec  5 14:07:49 2024 (6512304): Received 203 bytes data
Thu Dec  5 14:07:49 2024 (6512305): Checksum and header received ok!
Thu Dec  5 14:07:53 2024 (6516351): Lost MQTT connection!
`

geduxas, 2024-12-06:
I see in logs mqtt reconnect, but don't see mqtt reconect increase. So it's not what we are looking

julianher17, 2024-12-06:
Hey, 

Thanks for your feedback, but i don´t get it. What does that mean? There is no advanced log i could share?
After every Lost MQTT connection i see one increase in my heishamon dashboard.

Right now i have 2181 mqtt reconnects in 18hours uptime - correct received data 100%.
Heishamon V 3.8

geduxas, 2024-12-06:
Look to log, yes there is message about "```Lost MQTT connection!```" ( i think it false possitive, or bug in code) but truly it did'd. and in  "Heishamon stats" 

``` Heishamon stats: Uptime: 0 days 1 hour 48 minutes 31 seconds ## Free memory: 76% ## Heap fragmentation: 20% ## Max free block: 23088 bytes ## Free heap: 29272 bytes ## Wifi: 100% (RSSI: -50) ## Mqtt reconnects: 217 ## Correct data: 100.00% Rules active: 0```
it will keep in 217, so no increment here.. that's why i am telling you that log is wrong for this case, i need log where it increases.

geduxas, 2024-12-06:
also share log's from mqtt server

julianher17, 2024-12-06:
I understand, thanks for the explaination!

Here the new logs:

Fri Dec  6 15:05:17 2024 (60253): Heishamon stats: Uptime: 0 days 0 hours 1 minute 0 seconds ## Free memory: 66% ## Heap fragmentation: 7% ## Max free block: 23640 bytes ## Free heap: 25416 bytes ## Wifi: 82% (RSSI: -59) ## Mqtt reconnects: 2 ## Correct data: 100.00% Rules active: 0
Fri Dec  6 15:05:17 2024 (60273): Requesting new panasonic data
Fri Dec  6 15:05:17 2024 (60276): sent bytes: 111 including checksum value: 18 
Fri Dec  6 15:05:17 2024 (60283): Requesting new panasonic extra data
Fri Dec  6 15:05:17 2024 (60289): Already sending data. Buffering this send request
Fri Dec  6 15:05:17 2024 (60752): Received 203 bytes data
Fri Dec  6 15:05:17 2024 (60753): Checksum and header received ok!
Fri Dec  6 15:05:17 2024 (60754): received TOP1 Pump_Flow: 12.30
Fri Dec  6 15:05:17 2024 (60760): received TOP67 Compressor_Current: 7.8
Fri Dec  6 15:05:17 2024 (60767): received TOP115 Water_Pressure: 2.04
Fri Dec  6 15:05:17 2024 (60771): received TOP118 Second_Room_Thermostat_Temp: 127
Fri Dec  6 15:05:17 2024 (60779): Sending command from buffer
Fri Dec  6 15:05:17 2024 (60784): sent bytes: 111 including checksum value: 1 
Fri Dec  6 15:05:18 2024 (61259): Received 203 bytes data
Fri Dec  6 15:05:18 2024 (61259): Checksum and header received ok!
Fri Dec  6 15:05:18 2024 (61260): received XTOP0 Heat_Power_Consumption_Extra: 1838
Fri Dec  6 15:05:18 2024 (61267): received XTOP3 Heat_Power_Production_Extra: 6754
Fri Dec  6 15:05:22 2024 (65252): Lost MQTT connection!
Fri Dec  6 15:05:22 2024 (65252): Reconnecting to mqtt server ...
Fri Dec  6 15:05:22 2024 (65284): Heishamon stats: Uptime: 0 days 0 hours 1 minute 5 seconds ## Free memory: 71% ## Heap fragmentation: 13% ## Max free block: 23640 bytes ## Free heap: 27392 bytes ## Wifi: 82% (RSSI: -59) ## Mqtt reconnects: 3 ## Correct data: 100.00% Rules active: 0

geduxas, 2024-12-06:
Nothing wrong with your heishamon, something is on your network or mqtt server.. could you get logs from mqtt server?

julianher17, 2024-12-06:
Heres the logs from the mqtt server:

2024-12-06 16:07:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:07:52: New client connected from 192.168.178.150:59778 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:08:13: New connection from 192.168.178.196:63030 on port 1883.
2024-12-06 16:08:13: Client HeishaMon already connected, closing old connection.
2024-12-06 16:08:13: New client connected from 192.168.178.196:63030 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:08:22: New connection from 192.168.178.150:53694 on port 1883.
2024-12-06 16:08:22: Client HeishaMon already connected, closing old connection.
2024-12-06 16:08:22: New client connected from 192.168.178.150:53694 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:08:43: New connection from 192.168.178.196:57255 on port 1883.
2024-12-06 16:08:43: Client HeishaMon already connected, closing old connection.
2024-12-06 16:08:43: New client connected from 192.168.178.196:57255 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:08:52: New connection from 192.168.178.150:53239 on port 1883.
2024-12-06 16:08:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:08:52: New client connected from 192.168.178.150:53239 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:09:13: New connection from 192.168.178.196:64631 on port 1883.
2024-12-06 16:09:13: Client HeishaMon already connected, closing old connection.
2024-12-06 16:09:13: New client connected from 192.168.178.196:64631 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:09:22: New connection from 192.168.178.150:58050 on port 1883.
2024-12-06 16:09:22: Client HeishaMon already connected, closing old connection.
2024-12-06 16:09:22: New client connected from 192.168.178.150:58050 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:09:43: New connection from 192.168.178.196:53675 on port 1883.
2024-12-06 16:09:43: Client HeishaMon already connected, closing old connection.
2024-12-06 16:09:43: New client connected from 192.168.178.196:53675 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:09:52: New connection from 192.168.178.150:61104 on port 1883.
2024-12-06 16:09:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:09:52: New client connected from 192.168.178.150:61104 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:10:13: New connection from 192.168.178.196:54018 on port 1883.
2024-12-06 16:10:13: Client HeishaMon already connected, closing old connection.
2024-12-06 16:10:13: New client connected from 192.168.178.196:54018 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:10:22: New connection from 192.168.178.150:55110 on port 1883.
2024-12-06 16:10:22: Client HeishaMon already connected, closing old connection.
2024-12-06 16:10:22: New client connected from 192.168.178.150:55110 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:10:43: New connection from 192.168.178.196:64741 on port 1883.
2024-12-06 16:10:43: Client HeishaMon already connected, closing old connection.
2024-12-06 16:10:43: New client connected from 192.168.178.196:64741 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:10:52: New connection from 192.168.178.150:63369 on port 1883.
2024-12-06 16:10:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:10:52: New client connected from 192.168.178.150:63369 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:11:13: New connection from 192.168.178.196:61331 on port 1883.
2024-12-06 16:11:13: Client HeishaMon already connected, closing old connection.
2024-12-06 16:11:13: New client connected from 192.168.178.196:61331 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:11:22: New connection from 192.168.178.150:63060 on port 1883.
2024-12-06 16:11:22: Client HeishaMon already connected, closing old connection.
2024-12-06 16:11:22: New client connected from 192.168.178.150:63060 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:11:43: New connection from 192.168.178.196:51266 on port 1883.
2024-12-06 16:11:43: Client HeishaMon already connected, closing old connection.
2024-12-06 16:11:43: New client connected from 192.168.178.196:51266 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:11:52: New connection from 192.168.178.150:50899 on port 1883.
2024-12-06 16:11:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:11:52: New client connected from 192.168.178.150:50899 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:12:13: New connection from 192.168.178.196:49640 on port 1883.
2024-12-06 16:12:13: Client HeishaMon already connected, closing old connection.
2024-12-06 16:12:13: New client connected from 192.168.178.196:49640 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:12:22: New connection from 192.168.178.150:51491 on port 1883.
2024-12-06 16:12:22: Client HeishaMon already connected, closing old connection.
2024-12-06 16:12:22: New client connected from 192.168.178.150:51491 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:12:43: New connection from 192.168.178.196:52758 on port 1883.
2024-12-06 16:12:43: Client HeishaMon already connected, closing old connection.
2024-12-06 16:12:43: New client connected from 192.168.178.196:52758 as HeishaMon (p2, c1, k5, u'julian').
2024-12-06 16:12:52: New connection from 192.168.178.150:50809 on port 1883.
2024-12-06 16:12:52: Client HeishaMon already connected, closing old connection.
2024-12-06 16:12:52: New client connected from 192.168.178.150:50809 as HeishaMon (p2, c1, k5, u'julian').

julianher17, 2024-12-06:
I think I am stupid. I have to generate an own mqtt user per device?

geduxas, 2024-12-06:
> I think I am stupid. I have to generate an own mqtt user per device?

It's depends to server, i also use same user/password for all my mqtt devices.. i think it's identification problem, every hardware should identify with different name.

julianher17, 2024-12-06:
Nothing changed with different user.
There is no setting for an hardware ID or something?

I am using HomeAssistant with Mosquitto broker...

julianher17, 2024-12-06:
I think I got it now! If you are using multiple HeishaMons you have to change the HostName!
After using 2 different HostNames I got no reconnects.

Crazy... Thanks a lot for your help @geduxas

closed 2025-01-12 by github-actions[bot]