heishamon/HeishaMon #592: No data received after 2 days - Large PCB / long distance

DE-Kehrmaschine, 2024-11-26:
Hello everyone,
I have installed the large PCB with TTL-to-RS485 and 5V supply. This also worked for a few days. Since yesterday unfortunately not anymore.
Uptime on the web has been reset. No more data is being received.
Already tried:
-reboot
-firmware update
-TTL-to-RS485 swapped on the heat pump side. (no LED lights up here. normal?)
-The following is output by the console:

```
Tue Nov 26 11:52:20 2024 (315076): Heishamon stats: Uptime: 0 days 0 hours 5 minutes 15 seconds ## Free memory: 69% ## Free PSRAM: 2071876 bytes ## Free heap: 235796 bytes ## Wifi: -1% (RSSI: 0) ## Ethernet: connected (192.168.0.75) ## Mqtt reconnects: 1 ## Correct data: 0.00% Rules active: 0
Tue Nov 26 11:52:20 2024 (315078): Requesting new panasonic data
Tue Nov 26 11:52:20 2024 (315079): sent bytes: 111 including checksum value: 18 
Tue Nov 26 11:52:20 2024 (315080): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Nov 26 11:52:20 2024 (315080): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Nov 26 11:52:20 2024 (315081): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Nov 26 11:52:20 2024 (315082): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Nov 26 11:52:22 2024 (317084): Previous read data attempt failed due to timeout!
Tue Nov 26 11:52:22 2024 (317084): Received 0 bytes data
```

Maybe someone has an idea?

IgorYbema, 2024-11-26:
Do you see the tx led on the RS485 module on the heishamon side blink? And you say it doesn't blink at the heatpump side? Then there is something wrong with those modules or the line in between

DE-Kehrmaschine, 2024-11-26:
Yes, the LED blink on the Heishamon side, but not on the heatpump side. As I have already replaced the module there, I will test the line.
The module on the heat pump side should also light up without a cable to the heat pump, or does the module there get the 5V from the PCB of the heat pump?

IgorYbema, 2024-11-26:
It gets power from the heatpump. But you will not see any tx or rx led blink as long it doesn't receive anything from heishamon side. The heatpump will never 'talk' if it doesn't get a message from heishamon first. 
So order is:
-Heishamon sends request
-RS485 module heishamon side TX led blinks (or maybe RX led, not sure)
-RS485 module heatpump side RX led blinks (or again maybe TX)
-heatpump receives request
-heatpump sends answer
-RS485 module heatpump side TX led blinks (or again, maybe RX)
-RS485 module heishamon side RX led blinks (or again, maybe TX)
-finally heishamon receives the answer

IgorYbema, 2024-11-27:
So RX on the heishamon side. You say permanent. It should flicker each 5 seconds for a small burst, not stay on all the time.

As long as the heatpump side RS4585 isn't showing any TX it isn't; talking to the heatpump. So nothing with the heatpump PCB because it just doesn't get a message from heishamon. First thing is to make sure both RS485 modules are showing RX one one and TX on the other flashing.

DE-Kehrmaschine, 2024-12-01:
a bit annoying. It worked for 4 days. This time, however, the TX LED on the Heishamon side is flashing. Heatpump side does nothing. I will now try modules from another manufacturer.

5V supply (USB) works fine.

closed 2025-01-07 by github-actions[bot]