heishamon/HeishaMon #593: HeishaMon big one and Eth connetion problem

Udo3418, 2024-11-27:
I have installed a HeishaMon big one and can't get it to be addressed via the Eth interface. 
What do I have to set to switch the WiFi function to Eth? 
WiFi works problem-free so far.

IgorYbema, 2024-11-27:
It just works plug and play. It should get a IP address from your network automatically if you connect the ethernet cable to it. 
Does the module green and yellow leds work?

Udo3418, 2024-11-27:
No WiFi SSID and No Pw set.
Eth modul Led green and yellow blinking
on the same 2 port switch is the Panasonic Adapter working 
fritz.box can´t find new ip for heisamon
is this automaticly with Dhcp ?

Udo3418, 2024-11-27:
check this tomorrow it´s raining outside

Udo3418, 2024-11-28:
Power green
Status Led blue on not blinking
Eth Adapter yellow blinking

IgorYbema, 2024-12-03:
If your status LED is always BLUE this means that is it logging a bunch of data. One BLUE blink is a log line which is printed to the console.  Maybe check the logs for a clue what is happening? Maybe this is causing heishamon not getting an IP address

rvdgaag, 2024-12-07:
My best guess, issue caused by the limited "switching" capacity of the Fritzbox. Pull a cable to an individual port to the router.

closed 2025-01-12 by github-actions[bot]