heishamon/HeishaMon #595: Faulty Aquarea Inside Pipe Temperature

Ba-lu, 2024-12-04:
Hi,

I am using the large board with Version: 3.8. I also connected the CZ-TAW1 to the board.

In the last days I saw many faulty values for "Inside Pipe Temperature" which were below -10°C for some seconds. This resulted in the heating element running for 10 minutes with 3-4 kW. Over the course of the day I saw 20 occurences of the low temperatures. Before using heishamon I never had such values.

I am using the CN-NMODE cable and WiFi.

Wifi signal: 100%
Ethernet status: not connected
Memory free: 69%
Correct received data: 99%
MQTT reconnects: 13
Uptime: 41 days 17 hours 3 minutes 25 seconds

Currently I am not using an external 5V power supply.

geduxas, 2024-12-04:
I think it's due or defrost, and you have low temperature or flow rate... Heishamon can't do it.

Ba-lu, 2024-12-04:
It was around 0°C outside, the lowest temperature that was recoreded for inside pipe was -22°C. So defrosting should not result in such low recorded temperatures.

MiG-41, 2024-12-04:
It will. 
If you have split model you can try ro feel it by touching gas pipe (especially this smaller).
When real defrost starts ( when you hear the specific noise) this pipe momentally freeze.

Ba-lu, 2024-12-04:
I have a Panasonic WH-MDC07J3E5 (Monobloc), if the pipe temperature drops below 0°C the pipes will freeze and burst. The pipes run water.

I saw the temperatures over the day where there was no defrosting.

And I dont think the defrosting has to activate the heating element. At least not in my region.

MiG-41, 2024-12-04:
Yes , if the water in pipe will be below 0 it will freeze , that is why during defrost water pump speeds up to max , to bridge as much hot water from floor to Exchange heater as possible .

Also defrosts are confirmed , when outside temperatures are 4 or even 5deg.

geduxas, 2024-12-04:
Defrost wil switch pump in reverse mode, so it will Cool inside while heating outside unit.. so yes it will and it can! But yes pump have some antifreeze operations, to protect heat exchanger, and yes it will immediately start backup heater's. So it looks everything right. It's question why it's dropping so low, normally it shouldn't happen

Ba-lu, 2024-12-05:
There is nothing wrong with HeishaMon.

Today I saw when the defrost was happening and the temperatures for "inside pipe" dropped. So everything seems to be working as expected. I was not expecting that the temperatures drop below zero without freezing the water and bursting the pipe.

Please close the issue.

closed 2024-12-05 by Ba-lu

geduxas, 2024-12-05:
> There is nothing wrong with HeishaMon.
>
> Today I saw when the defrost was happening and the temperatures for "inside pipe" dropped. So everything seems to be working as expected. I was not expecting [...]

It's not freezing because of flow and pressure.