heishamon/HeishaMon #599: Loading the rules fails with "ERROR: unknown token"

McMagellan, 2024-12-07:
This example rulesset cannot be loaded. I have reduced them to a minimum.
Tested with version alpha 5d79138 (7.12.2014) on Heishamon V4.

```
on DeltaCheck then
  #Delta = #Delta + 1;
end

on @Main_Outlet_Temp  then
  DeltaCheck();
end

```

If I swap the two rules, the ruleset can be loaded successfully.
```

on @Main_Outlet_Temp  then
  DeltaCheck();
end

on DeltaCheck then
  #Delta = #Delta + 1;
end
```

In the event of an error, the following can be seen in the console.

[image]

Normally the ruleset starts with the system trigger on System#Boot and the error does not occur.
But if I only partially load individual groups to isolate an error, this situation occurs.

I think it's because the first function handle in the list cannot be called from the following rules.

CurlyMoo, 2024-12-09:
Can you post a link to the build or commit the version you tested?

McMagellan, 2024-12-10:
This behavior is also included in version 3.8. I specifically used the latest test version to see if this bug still exists.

[image]

CurlyMoo, 2024-12-10:
To be honest, I have no clue why this happens.

Read with me.

The only place that triggers this error is this:
https://github.com/IgorYbema/HeishaMon/blob/3dbd4ca3572c12279092ad488219d21ff7ff29b7/HeishaMon/src/rules/rules.cpp#L1322-L1326

The `is_variable` function is one of the functions that can prevent this `else` statement to be reached:
https://github.com/IgorYbema/HeishaMon/blob/3dbd4ca3572c12279092ad488219d21ff7ff29b7/HeishaMon/src/rules/rules.cpp#L1161

In case of an `#`, the `is_variable` doesn't do anything with it except returning it's length:
https://github.com/IgorYbema/HeishaMon/blob/3dbd4ca3572c12279092ad488219d21ff7ff29b7/HeishaMon/rules.cpp#L121-L123

https://github.com/IgorYbema/HeishaMon/blob/3dbd4ca3572c12279092ad488219d21ff7ff29b7/HeishaMon/rules.cpp#L224

CurlyMoo, 2024-12-10:
Does the behavior change when you load this:
```
on DeltaCheck then
  #Delta = 1;
end

on @Main_Outlet_Temp  then
  DeltaCheck();
end
```

McMagellan, 2024-12-10:
The same error happens with your example.

I could imagine the following cause:
During the parsing process, the individual rules are listed, inventoried and given address numbers, e.g. #0, #1, #2 and so on.
If a reference to a previously inventoried rule has to be inserted in one of the following rules, this will not work if the link has to point to the rule in #0.

CurlyMoo, 2024-12-11:
That could be true in a later stage of parsing. At the point where the error is triggered nothing more happens then just checking if the variable format is valid. Which is done in the linked `is_variable` function.

closed 2025-01-17 by github-actions[bot]