heishamon/HeishaMon #601: Heishamon large parallel operation with CZ-TAW1

rasy35b, 2024-12-09:
I have purchased on Tindie Heishamon PCB large with ETH module and CM-CNT cable.  Software instated on HM is 3.8. I used original cable to connect HP Pana to HM and this bought cable to connect HM to CZ-TAW,1 ETH connected to router and external power supply to USB C port.  I can operate (change settings on HP) both way from HM and from Panasonic cloud via CZ-TAW1. So, for instance I can set from Homeassistant Force_DHW_State ON and from Pana cloud I can set also Force DHW state as ON . Is this correct? I have tried to set configuration HM on listening mode but couldn't make this configuration operational and I set on "Enable cztaw proxy port".

[image]
[image]

blb4github, 2024-12-13:
You can both monitor and control the HP from HeishaMon and CZ-TAW1 if the CZ-TAW1 is connected to HeishaMon proxy port. That's the beauty of this! As far as I know Heisamon is not doing any manipulation of messages between HP and CZ-TAW1.

closed 2025-01-19 by github-actions[bot]

hendrer, 2025-01-19:
I am planning on taking this route - what cable do you need from HM Large to CZ-TAW1 for proxy? I am using original cable from HP to HM (small currently, large in the future) currently

rasy35b, 2025-01-19:
For me works CN-CNT cable .

hendrer, 2025-01-19:
ok so HM -> CZ-TAW1 with CN-CNT and from HP -> HM with CN-NMODE? Thanks for the quick reaction!

rasy35b, 2025-01-19:
My connection is HP to HM by CN-CNT and HM to CZ-TAW1 original from CZ-TAW1 set. Proxy works perfectly.

IgorYbema, 2025-01-19:
HM to CZ-TAW1 must be CN-CNT cable.
Form HM to HP you can use CN-CNT or CN-NMODE
Original cable is same as CN-CNT cable.

Only if you plan to use the small heishamon (without the extra port to CZTAW1), you must use the CN-NMODE if your are planning to install it together with CZTAW1, as that one is already connected to HP with original cable). And then run listen only mode because only one can be directly connected to HP.