heishamon/HeishaMon #602: Add possibility to switch relay state via rules

NerdyProjects, 2024-12-10:
I (want to) use the relay to tell an external heating system controller to turn on the pumps / set the mixing valves and to not rely on another computer, I would like that to happen via rules on heishamon.

Roelaynd, 2024-12-12:
I concur, I would like to have control over the relays via the rules as well for setting a valve instead of having to rely on Home Assistant.

(Great job thus far btw, Heishamon can't be praised enough)

blb4github, 2024-12-13:
This would be a great feature!!

Djongambo, 2024-12-14:
I would also highly appreciate that to control an additional external pump. Need to switch that pump off when heatpump goes into defrost mode.

joep2000, 2025-01-08:
+1
the M (with only remote) has no contact for extra pump or heating element or whatever. Heishamon relays in rules would be perfect option!

IgorYbema, 2025-01-13:
is added in this commit https://github.com/IgorYbema/HeishaMon/commit/47b4855249ca7f6b281b99a1724b81500e7e379b
will be available in v3.9 which will be released soon

closed 2025-02-19 by github-actions[bot]

blb4github, 2025-02-19:
To my knowledge this is implemented in 3.9