heishamon/HeishaMon #604: M Series

earburner, 2024-12-13:
Hi all, i am playing with my heishamon and my new M series heat pump, its very new and i know i was one of the first to have the Internal CME8 unit which is why the output from heishamon isn't that great :smile: was wondering if i can add to the project and and maybe dump some output for some decoding :slightly_smiling_face: heishamon large board v3.8

not sure how i can get a dump out of the board (ssh etc) but i can also get a dump out of the Panasonic service cloud to help with the decoding of the data

MiG-41, 2024-12-13:
Generally , if you see some information/data  in service cloud , with HeishaMon is not providing then we can analize some logs to have this datas also in Heishamon.

earburner, 2024-12-13:
there are 2 CN-CNT ports on the M Series so data can go to both service cloud and Heishamon.

geduxas, 2024-12-13:
@MiG-41 this is hydro-split unit which can operate standalone  (as monoblock and with limited capabilities) and inside unit just extends it's capabilities to normal unit.

closed 2025-01-19 by github-actions[bot]

ReinhardGruber, 2025-04-10:
> Hi all, i am playing with my heishamon and my new M series heat pump, its very new and i know i was one of the first to have the Internal CME8 unit which is why the output from heishamon isn't that great 😄 was wondering if i can add to the project and and maybe dump some output for some decoding 🙂 heishamon large board v3.8
> 
> not sure how i can get a dump out of the board (ssh etc) but i can also get a dump out of the Panasonic service cloud to help with the decoding of the data

hi, did you found a solution?

MiG-41, 2025-04-10:
Solution for ?

ReinhardGruber, 2025-04-10:
> Sie erhalten nicht häufig E-Mails von ***@***.*** Erfahren Sie, warum dies wichtig ist<https://aka.ms/LearnAboutSenderIdentification>

using HeishaMon with the new Aquarea M series

MiG-41, 2025-04-10:
Who said that it is not working ?

ReinhardGruber, 2025-04-10:
I could not find the pumpname in the file with the pump ids. So i thougth it isn't working. May i'm wrong. Can you confirm?

earburner, 2025-04-10:
> hi, did you found a solution?

After the last firmware, it has been working well