heishamon/HeishaMon #605: (Almost) No Data received from HP

Mc-fry, 2024-12-17:
Hi guys... I have the following Setup.
ESP8266 with Heishamon and self soldered Level shifter circuit like the linked one in readme File.
Self made shielded cable in cn-nmode connector of J series Heatpump. 

Heishamon WiFi and Webinterface are working. But i  almost never get Data from heatpump. After reboot maybe one or two times it works but then I always get the "0 bytes received" error. I let it run several days, no more data.

Things I tried to troubleshoot:
- Checked Rx tx polarity several times
- measured my wire connectivity
- powered esp on lab bench and created loopback with additional pullup. it received its own data without problem (level shifter works)
- checked heatpump 5v (ok)

What could be the problem here? Is my heatpump transmit circuit broken? Does it lock out because I try to communicate every 5 sec?
Why do rebooting or plugging/unplugging cn-nmode sometimes help for one or two data packages? 

Maybe someone can help. Thx!!!

Here is a screenshot of the console
[image]

[image]

MiG-41, 2024-12-24:
For sure bad connection.
Maybe even bad soldering on cn-cntmode socket on PCB ( but that is the last thing).

IgorYbema, 2024-12-24:
What esp8266 board are you using? Did you add the gpio5 enable line?

Mc-fry, 2025-01-05:
Thank you for your replies. I got it to work in the meantime. It was a very odd problem, let me explain.
I borrowed a simple scope from a colleague and checked the waveforms. Transmit to the heatpump was fine. The reply from the heatpump was there, but the low level of the serial did not go to ground but about 2V. So the esp did not see the data. Maybe it was just at the threshold so that it worked rarely. This might be indeed a bad solder joint on the heatpump PCB, maybe a resistance in the ground connection...
So what I did is to use a lm311 comparator instead of the original tx level shifting mosfet. I soldered a poti in to set the threshold for logic low. It immediately worked and I have 100% correct data since 2 weeks.

Mc-fry, 2025-01-05:
So this is closed now.

closed 2025-01-05 by Mc-fry