heishamon/HeishaMon #606: Long rules sets not accepted with FATAL: Internal error in rule_run #4401

blb4github, 2024-12-18:
I'm experiencing problems with long rules sets. attached 3 rules sets:

The rules set 20241218o_rc can be loaded:
[Heishamon_rules20241218o_rc.txt](https://github.com/user-attachments/files/18187433/Heishamon_rules20241218o_rc.txt)

The rules set 20241218p_rc is the same as 20241218o_rc but with some additional logic in TaShift:

```
if #allowHeatDelta == 1 then
	$HD = -1;
	if #CompRunTime > -1 && #CompRunTime < 3 then
		$HD = #HeatDelta + 5;
	elseif #CompRunTime > 20 && #CompRunTime < 25 then
		$HD = #HeatDelta;
	end
	if @Heat_Delta != $HD && $HD > 0 then @SetFloorHeatDelta = $HD;end
end
```

this rules set is not accepted, it ends with error:
```
rule #8 was executed in 1355 microseconds
bytecode: 404/404, heap: 80/80, stack: 4/84 bytes, varstack: 1032/1048 bytes
rule #9 was prepared in 22656 microseconds
rule #9 bytecode was created in 36250 microseconds
bytecode: 212/212, heap: 52/52, stack: 4/84 bytes, varstack: 1072/1088 bytes
FATAL: Internal error in rule_run #4401
```

[Heishamon_rules20241218p_rc.txt](https://github.com/user-attachments/files/18187434/Heishamon_rules20241218p_rc.txt)

If I take out some lines in rule `syncOT`, completely not related to the addition above the rules set is accepted again, the lines with -- in front are taken out:
```
		if #CompState > 0 then
			?flameState = 1;
			if @Heat_Power_Consumption > 0 then ?chState = 1; else ?chState = 0; end
--			if @DHW_Power_Consumption > 0 then ?dhwState = 1; else ?dhwState = 0; end
--			if @Cool_Power_Consumption > 0 then ?coolingState = 1; else ?coolingState = 0; end
		else
			?flameState = 0;
			?chState = 0;
--			?dhwState = 0;
--			?coolingState = 0;
		end
```

[Heishamon_rules20241218q_rc.txt](https://github.com/user-attachments/files/18187435/Heishamon_rules20241218q_rc.txt)

Binifada1956, 2024-12-19:
already tested splitting TaShift again?

blb4github, 2024-12-19:
yes, I tested it as well with TaShift splitted, it gives the same error:

[Heishamon_rules20241219b_rc.txt](https://github.com/user-attachments/files/18202671/Heishamon_rules20241219b_rc.txt)

```
rule #9 was prepared in 181044 microseconds
rule #9 bytecode was created in 249726 microseconds
bytecode: 404/404, heap: 80/80, stack: 4/84 bytes, varstack: 1032/1048 bytes
FATAL: Internal error in rule_run #4401
```

closed 2025-02-27 by github-actions[bot]