heishamon/HeishaMon #607: Compatibility L Series 5kW Model ODU:WH-WDG05LE5

stuxnetdll, 2024-12-21:
I have been running heishamon with a IDU:WH-SDC0509L3E5 ODU:WH-WDG0**5**LE5 for the past few days successfully. However, I noticed that the 5kW unit is not listed as a supported model leading to TOP92 Heat_Pump_Model showing as 255 unknown. It appears that all data is shown just fine and using the HA integration, I am able to see and control the unit. The only limitation here is that due to the HA integration not recognizing it as an L series unit, I am unable to access the L series specific variables in HA.
As I can successfully integrate the unit in HA and everything else appears to work fine otherwise, I suppose that the device is likely fully compatible and has just not been configured in heishamon as one of the supported models.
Are there any plans to include this model as well? If any data dumps etc. would be required from my 5kW model to achieve this, I would be happy to support.

IgorYbema, 2024-12-21:
The TOP92 model is just a hard coded thing. Doesn't change the workings of heishamon. if it says 'uknown'. It gets all the data it has.
I'm not sure if the HA component does make a difference if it detects a L-series but that is out of my scope. Heishamon has all the data and sends this to HA.

geduxas, 2024-12-22:
> The TOP92 model is just a hard coded thing. Doesn't change the workings of heishamon. if it says 'uknown'. It gets all the data it has. I'm not sure if the HA component does make a difference if it detects a L-series but that is out of my scope. Heishamon has all the data and sends this to HA.

I think it's need to get out of that model number thing.. it will not be possible to hardcode every model and revision of that :D

IgorYbema, 2024-12-22:
yes, maybe just remove top92

geduxas, 2024-12-22:
I think it's better to split this byte to separate topics (as panasonic do) and just leave it.

IgorYbema, 2024-12-22:
do you have a quick example on how panasonic stores in in the menu? I believe you are quicker to have this example then me :)

geduxas, 2024-12-22:
> do you have a quick example on how panasonic stores in in the menu? I believe you are quicker to have this example then me :)

Here it is : 

https://github.com/Egyras/HeishaMon/issues/189#issuecomment-971744190

geduxas, 2024-12-22:
For example my model:
E2 CF 0C 74 09 12 D0 0D 95 05  (12kW t-cap)
Will be:
CFE1
09740B
D011
05950C

One for IDU, another for ODU (boards), don't know exactly what would those numbers should mean. It could be main PCB type/configuration.. 0B and 0C is definitely some kind of revision.. (rev.1 0A, rev 2 0B..)

IgorYbema, 2024-12-22:
Yes if I take the current known list and remove those two bytes (3rd and 8th) all models which have 'new model' as marking already exists before. So that is good thing. We can make the list a bit shorter now.

closed 2025-01-28 by github-actions[bot]