heishamon/HeishaMon #608: Parsing Error in iobroker --> stats file

McMagellan, 2024-12-25:
I'm using Heishamon V4 with firmware #664

[image]

Constant errors (cannot parse ...) are displayed in the iobroker log

[image]

To me it looks like the value of the parameter "board" was put in double quotation marks when composing the JSON string.

Any news on using Heishamon V5 relays from within Rules or with URL command?
Or just decoration?

MiG-41, 2024-12-25:
parameter "version" don't have closing quotation mark sign , so after "Alpha-d05c827 should be " ... 
@IgorYbema , possible to fix it and give new actual Alpha ?

MiG-41, 2025-01-01:
@IgorYbema , do you have compiled some new Alpha with fixing this ?
I'm also parsing from this topic HeishaMon MQTT reconnect no. and becouse of this , HA also don't recognize it as JSON and fail.