heishamon/HeishaMon #609: TOP44 @Error cannot be read in rules

McMagellan, 2024-12-25:
I tried to process an error code that Pana reported via TOP44 in rules. 
The trigger on @Error works as it should.
But when I want to access the value I always get the value 0 back even though TOP44 contains the error code e.g. F36.

It is also strange that the TOP44, in contrast to all other TOPs, has to be treated as a string because this value is the only one that is not numeric but also contains a letter.

Or could it be that Rules also has a variable called Error internally and that a query refers to the internal variable?

Ping20002015, 2024-12-28:
Same problem here!

CurlyMoo, 2024-12-30:
Retrieving values from TOP's that contain string values was not yet supported. Will look into it later.

CurlyMoo, 2024-12-30:
Should be fixed in: https://github.com/IgorYbema/HeishaMon/pull/145

closed 2025-01-16 by McMagellan