heishamon/HeishaMon #612: Should Leisha be known to Heishamon?

klausj1, 2024-12-31:
I have a Leisha connected to my Heishamon, and the Heishamon states:
`TOP92 | Heat_Pump_Model | 255 | unknown`

Should that work already?

The reason is that in a heatpump-forum I was asked to provide L-series-specific temperatures, and as the Leisha is not even recognized, I doubt that there are such L-series-specific temperatures.

My exact model is a WH-WDG05LE5 (outside) and WH-SDC0509L3E5 (inside).

geduxas, 2024-12-31:
What is "Leisha'?
Heatpump model don't do anything special for all work. It just "string" and nothing else, if your model is not recognized it just means that your model name is not coded in heishamon, and everything else will work as is.

klausj1, 2024-12-31:
Leisha is the L-series of the Panasonic Aquarea.

thecem, 2025-01-04:
```
  0  1  2  3  4  5  6  7  8  9  10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31
0 71 C8 01 10 56 55 52 49 00 55 00 15 00 00 00 00 00 00 00 00 59 15 11 55 56 15 55 55 55 19 56 00 
1 00 00 00 00 00 00 7E 80 80 80 B4 71 71 FF 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
2 D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A3 99 7B 8F B7 A3 7B 8F 92 85 7B 8F 8A 94 9E 8F 8A 94 9E 85 8F 
3 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 75 55 21 A1 15 59 05 13 13 66 21 51 15 01 51 9C F0 
4 00 E2 D5 0C 67 00 83 92 0C 27 98 A0 00 98 7F 9B A0 32 32 9F B3 32 32 32 7E 9F A1 A9 98 9A 7B 00 
5 7C 00 00 37 01 1A 2E 00 00 52 11 26 51 38 01 49 01 01 01 82 02 00 6A 06 00 01 00 00 01 00 00 06 
6 01 00 00 00 00 00 00 04 80 24 D9 
```

Where 4 -1 up to 4-10 is Byte 129 to 138:

**E2 D5 0C 67 00 83 92 0C 27 98**

Indoor: WH-ADC0509L3E5AN
Outdoor: WH-WDG05LE5
KIT: KIT-WC09L3E5

PR:
https://github.com/Egyras/HeishaMon/pull/614

klausj1, 2025-01-04:
Thx! I have a different Indoor model: WH-SDC0509L3E5. Is this included in your change as well?

thecem, 2025-01-04:
pls send me the hex dump out of the heishamon console, then I could check this or you.

klausj1, 2025-01-04:
Like this?

Sat Jan  4 11:15:14 2025 (30501): sent bytes: 111 including checksum value: 1 
Sat Jan  4 11:15:14 2025 (30502): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:14 2025 (30503): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:14 2025 (30504): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:14 2025 (30504): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30886): Received 203 bytes data
Sat Jan  4 11:15:15 2025 (30887): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 88 00 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 
Sat Jan  4 11:15:15 2025 (30888): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30889): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30890): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30890): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30891): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jan  4 11:15:15 2025 (30892): data: 00 00 00 00 00 00 00 00 00 00 9C 
Sat Jan  4 11:15:15 2025 (30892): Checksum and header received ok!

thecem, 2025-01-04:
No not the empty block, there is also a block with less Zero‘s.

klausj1, 2025-01-04:
Ah, seems to be E2 D5 0B 34 99 83 92 0C 27 98:

Sat Jan  4 11:23:40 2025 (535982): data: 71 C8 01 10 65 95 62 49 00 55 00 15 00 00 00 00 00 00 00 00 69 15 11 55 56 25 55 55 55 19 59 00 
Sat Jan  4 11:23:40 2025 (535983): data: 00 00 00 00 00 00 80 80 80 80 AF 71 71 B1 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Sat Jan  4 11:23:40 2025 (535984): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AA 9D 76 8F B7 A3 7B 8F 8E 83 76 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sat Jan  4 11:23:40 2025 (535985): data: 8A 0D 5B 7A C1 0B 7E 7C 1F 7C 7E 00 00 00 55 75 55 21 87 15 59 05 12 12 65 21 51 15 01 93 98 FE 
Sat Jan  4 11:23:40 2025 (535987): data: 00 E2 D5 0B 34 99 83 92 0C 27 98 97 00 AC 83 97 97 32 32 A4 AF 32 32 32 80 A4 99 91 94 8B 83 00 
Sat Jan  4 11:23:40 2025 (535988): data: 8C 00 00 18 01 01 01 00 00 D2 13 60 B1 01 01 79 01 01 01 EE 02 00 03 06 00 01 00 00 01 00 00 06 
Sat Jan  4 11:23:40 2025 (535999): data: 01 00 00 00 00 00 00 04 80 17 66

MiG-41, 2025-01-04:
> Thx! I have a different Indoor model: WH-SDC0509L3E5. Is this included in your change as well?

You have different outdoor unit as well ! So it is totally different kit also....

thecem, 2025-01-04:
no this is not included, d you have the kit number or the productnumber ?

klausj1, 2025-01-04:
I have a WH-WDG05LE5 (outside) and WH-SDC0509L3E5 (inside). The indoor unit is without a tank. Is this what you need?

thecem, 2025-01-04:
KIT number is the same as mine. I added the model combination 52 to the PR´s above. 
So we have to wait until egyras is approving and releasing a new version.

MiG-41, 2025-01-04:
@thecem , i assume this AN means , that this All-In-One is equipped with Anode. 
Can you do some logs enabling /disabling it , and perhaps trigger error on it ? https://github.com/Egyras/HeishaMon/discussions/611

thecem, 2025-01-04:
AN is the Version with DHW Boiler as all in One with Electrical Anode, there is also a version without.

L is the ODU Type (or Leisha the typname in german forum) Monoblock

If you could tell what to do, I could do and help here out : Can you do some logs enabling /disabling it , and perhaps trigger error on it ? https://github.com/Egyras/HeishaMon/discussions/611

MiG-41, 2025-01-04:
Please find in menu options for Anode ( if any) , and change it , when you change give a log ,and so on. We have to find with byte is responsible /changing ....

thecem, 2025-01-04:
Will do asap I have access to the IDU, Should be next weekend.

thecem, 2025-02-04:
Not stale….

IgorYbema, 2025-02-04:
Since v3.9 heishamon doesn't care about model name anymore. It is only confusing like this post. It doesn't make heishamon work better or worse if it didn't understand the model. L-series are working without any problem. The only discusison above was how to document the model number. So it is stale.

klausj1, 2025-02-04:
OK, closing this issue.

closed 2025-02-04 by klausj1