heishamon/HeishaMon #613: Wiring large board

lucasimons, 2025-01-02:
I see that my heat pump have the cn-nmode 4pin and cn-cnt 5pin 

The cable have 5pin so the only way to connect to the heat pump is 5pin cn-cnt and connect to heishamon board in c taw 5pin port as it is showed in this picture 

[image]

Is it correct?

MiG-41, 2025-01-02:
No !
On Heishamon PCB you have labels ! You have to connect wire to socket labeled HEATPUMP !

lucasimons, 2025-01-02:
Now it is clear because i see the board  connector and the heat pump conector, the clear way to explain in the description on tindie is cable for cz taw (my cable) and cable for ONLY heishamon board.... I have another question about 12v cable... The large board is power from 12v of HP?

IgorYbema, 2025-01-02:
12v isn't used, only the 5v is used to power heishamon. That is why the 4-wire cn-nmode cable works fine also.

Maybe a future heishamon board will use the 12v. But no plans currently.

lucasimons, 2025-01-02:
Perfect so now I can use this cable? Can I connect to HP pcb the cn-cnt 5pin cable connector and cut the other connector to solder the wire to heishamon pcb?

MiG-41, 2025-01-03:
Why you want to cut it ? Why solder to heishamon pcb ,since you have a ready connector ? Only thing you should do is to ( acc to the picture you attched ) connect free end to Heishamon socked called HEATPUMP , unplug from Heishamon CZ-TAW1 socket , and this one connect to HP CN-CNT...

IgorYbema, 2025-01-03:
You are talking about the CN-NMODE connector. That has 4 pins yes. But you don't need to use that connector. You have the CN-CNT cable, so you connect to the CN-CNT connector on the heatpump (which has 5 pins)

lucasimons, 2025-01-03:
> You are talking about the CN-NMODE connector. That has 4 pins yes. But you don't need to use that connector. You have the CN-CNT cable, so you connect to the CN-CNT connector on the heatpump (which [...]

yes, i write wrong in previus reply, so HP side i must use the CN-CNT 5pin, but heishamon side i must connect to heatpup socket that have the same plug o HP side, so i need to buy onother cable or cut the cable and solder to the bottom of heishamon pcb .... is that correct? 

if i understood i CAN'T use the cz taw socket on heisha pcb because it isn t in parallel with HP socket

IgorYbema, 2025-01-03:
> 
> yes, i write wrong in previus reply, so HP side i must use the CN-CNT 5pin, but heishamon side i must connect to heatpup socket that have the same plug of HP side, so i need to buy onother cable or cut the cable and solder to the bottom of heishamon pcb .... is that correct?

No. The heishamon port is a different 5 port connector. A bit larger. The left one. 
The other smaller port on the heishamon is the same indeed. You connect the CZ-TAW to this port.

> if i understood i CAN'T use the cz taw socket on heisha pcb because it isn t in parallel with HP socket

The CZ-TAW socket is there to connect the CZ-TAW to (the original panasonic internet module).

IgorYbema, 2025-01-03:
So like this
[image]

lucasimons, 2025-01-03:
> So like this [...]

the issue is that i DON'T have the orginal panasonic internet module, i ONLY have the heishamon large and the 5pin cable in the picture, how can i connect the board to HP? i need new cable?

IgorYbema, 2025-01-03:
No you can just use the cable you have in your hand on the picture. I really don't get it why you are not understanding this.
Do as what @MiG-41 said.
Starting from the photo you made:
- Unplug the cable from the small connector on the heishamon board. Plug that end on the CN-CNT connector on the heatpump
- Take the other end of the cable (the connector which you are holding in your hand) and connect that to the larger (left) connector on heishamon (marked heatpump)

It isn't simpeler than that!

If you don't have the original internet module forget everything about CZ-TAW etc. Just connect heishamon to the heatpump and go!

lucasimons, 2025-01-03:
sorry for wasting your time but I didn't see that the connectors on the Heishamon and HP side were different, now I'll connect everything

closed 2025-01-03 by lucasimons