heishamon/HeishaMon #616: No more than 6 Dallas DS18B20 sensors usable on the "small" (ESP8266) unit v4.7

Tool-use-today, 2025-01-04:
How many temp sensors are you normaly using on the "small" (ESP8266) unit v4.7? 

Before I used 11 sensors with one simple pull up around 4kOhm for all of them on an Wemos with ESPEASY installed on it. Max distance to the last sensor is approx 3m. These setting just worked fine.

Now I put the 11 sensors to the HeishaMon Board.

The same sensors will not be able to be read when I connect all of them to the unit. In fact no more than 6 pcs are running. If I only use 6 sensors all values are measured fine. But if I add one more sensor in the chain all of them will not be read anymore. Heishamon itself runs meanwhile without any problems. Only the temp sensors will not be read.

I tried knew sensors. Also I tried a flying circuit with knew ones. I also measured the volltate on the signal pin with an oscilloscope (of course I checked the supply pin as well). Somehow the levels between "no signal" and "signal" is not as expected between 3.3 (or more) an 0 volts. The signal does not reach a level more than 2.xx volts so the sensors seem not to work proberly. 

Maybe there is some weakness in the supply for the sensors or the pullup? Therefore I was wandering how many sensors are stable in use (proofed) and if someone had similar issues about the amount of connected sensors?

If someone would able to send me the shematics of the v4.7 unit I maybe could do some research then with the power supply of the sensors, also the pullup. Normally there should no reason why only these less amount of sensors should work on the board.

geduxas, 2025-01-04:
I used to use 8 sensor's, but all was around 1.5m-2m. didn't notice any problems.

Tool-use-today, 2025-01-04:
thx. Good to know

IgorYbema, 2025-01-04:
I have 11 installed myself on a small heishamon.
1 wire schematic is this. Very simpel. Maybe you fed the data line with 5v on your wemos network before (which some people do). This will make it a bit stronger but you might overload the gpio input on the esp8266. So that is why the heishamon has the data line on 3.3v (and power on 5v)

[image]

Tool-use-today, 2025-01-04:
Thank you as well. 4,7k on 3.3 Vcc should be enough though. Good to know that the board ist able to handle more sensors. I recheck my installation. Btw. do you have a source for the schematic?

IgorYbema, 2025-01-04:
I don't share the whole schematic but this is the relevant part for the 1wire

Tool-use-today, 2025-01-05:
ok, thank you for that. I am gonig to try a whole new setting with new wiring and sensors. This will take some time cause meanwhile I use some heishamon temperatures readings instead of my own ones. I will give a feedback if neccessary.

As others don´t have that issue it must depend on my installation. Therefore I close the case.

closed 2025-01-05 by Tool-use-today