heishamon/HeishaMon #619: declared variable with values below 0 are not exact

blb4github, 2025-01-05:
Values below 0 are not precise, e.g.:

declaring a variable like:

#PIDKi = 0.001;

give on the console:
#PIDKi = 0.000999998

CurlyMoo, 2025-01-23:
> It is important to realize integers are stored in 24 bits and floats in 27 bits. This still allows for big integers and high float precision, but less than in a regular 32bit enviroment. The reason for this is to allow for storing integers and floats in a 4 byte struct, where the first 5 bits of the first byte holds the variable type. The last three bytes store the actual number. For floats, the last 3 bits of the first byte are used to also store float to allow for higher precision.

https://github.com/CurlyMoo/rules/

blb4github, 2025-01-23:
I don't have full understanding of your explanation but I believe the conclusion is this is as designed and doesn't affect calculations (except the result of some calculations will be a bit less precise). Right?

CurlyMoo, 2025-01-23:
> I don't have full understanding of your explanation but I believe the conclusion is this is as designed

The amount of bits in which you can store a number defines the range it can take. And for floats, that indeed affects the presicion. That's also why we moved from 32bit to 64bit systems a few decades ago. In this case i even lowered 32bits to 27bits. So, the lack of precision is indeed as designed.

> and doesn't affect calculations 

It does, because calculating with less precise numbers give you a less exact result.

I believe the precision is good enough for usage in the HeishaMon.

A simply storing a variable in 4 byte instead of 8 bytes uses twice as less memory.

blb4github, 2025-01-23:
As designed and not problematic.

closed 2025-01-23 by blb4github

CurlyMoo, 2025-01-24:
And for the record. This doesn't only apply to negative numbers. This applies to all numbers with decimals.