heishamon/HeishaMon #625: Tasmota ?

mmmmm009, 2025-01-13:
Hi,

Juste wondering if it would be possible to write a driver for tamota?

geduxas, 2025-01-13:
I think, but is it worth it?

mmmmm009, 2025-01-13:
Tasmota is very convenient.
You flash it online.
So no additional software needed.
In Berry script you could configure all mqtt topics as you wish,
you could write your own "automations"/rules, 
all of it is available out of the box.
would be available to more people i guess

geduxas, 2025-01-13:
There is enough workload done in esp side to emulate and translate everything to mqtt.. so all mqtt endpoints are made, also there is some scripting capabilities... And hardware is near it's limits.. i think it's possible to switch to esp32, but still don't know how tasmota could bring more potential for this project... It's already do everything what it's need to do..

IgorYbema, 2025-01-13:
Tasmote and ESPhome are indeed possible but I don't think you find some here to put time in that.

ElVit, 2025-01-30:
Take a look at my repo [esphome_components](https://github.com/ElVit/esphome_components).
I'm working on an ESPHome Component based on Heishamon.
You can also find an [example yaml file](https://github.com/ElVit/esphome_components/blob/main/example_panasonic_heatpump.yaml) there.
Currently it is only possible to read values. Sending commands is not possible but I'm working on it.
So currently you have to connect the CZ-TAW1 (which will send you the requests) or you must send the requests manually to get some data from the heatpump.

closed 2025-03-08 by github-actions[bot]