heishamon/HeishaMon #628: SetCurves

robert-b, 2025-01-17:
Recently there has been a fix applied for SetCurves: https://github.com/Egyras/HeishaMon/issues/291
Unfortunetly it still does not work.
The described json document is wrong and fixing this has not solved the issue.

geduxas, 2025-01-17:
it's not in released firmware yet.  so you should need to wait, or test alpha firmware..

robert-b, 2025-01-17:
then it is time to test - as i need this functionality.
thanks for your help.

robert-b, 2025-01-17:
surprise

stumbaumr, 2025-01-17:
As far as I understand modifying the heat curves already work since release 3.8.
Sample from [my rules](http://aquarea.smallsolutions.de/index.php?title=Anlage_stumi75#F.C3.BCr_HeishaMon_zum_Hochladen):
`      @SetCurves = concat('{zone1:{heat:{target:{high:', $targetAfterDHW ,',low:', $targetAfterDHW ,'}}}}');`

Or am I missing something here?

robert-b, 2025-01-17:
i have no idea - but i can test the snippet.
up to now - nothing has been accepted when saving my rules.

stumbaumr, 2025-01-17:
You can test my rules... the snippet alone will not work...

robert-b, 2025-01-17:
ok - this snippet is bad:
@SetCurves = concat('{"zone1":{"heat":{"target":"high":', @Z1_Heat_Curve_Target_High_Temp, ',"low":', @Z1_Heat_Curve_Target_Low_Temp, '}}}}');

but this:
       @SetCurves = concat('{zone1:{heat:{target:{high:', $target ,',low:', $low ,'}}}}');
works
already tested.
my updated script works.
very interesting and thanks a lot.

stumbaumr, 2025-01-17:
So you should be able to close the issue then, right?

McMagellan, 2025-01-18:
I can confirm that SetCurves used in Rules has worked very well without any problems since 3.8.
For example: @SetCurves = concat('{zone1:{heat:{target:{high:',#VLRef,',low:',#VLRef,'}}}}');
Thanks for the implementation.

Are there any considerations about setting up a "Rules Wiki" where you can look up small code examples?

robert-b, 2025-01-20:
i can't close the issue because there is a difference between the actual implementation and the documentation.
both have to be in sync.

closed 2025-02-25 by github-actions[bot]