heishamon/HeishaMon #632: Heishamon crashes after IP-Scan

naseweiss99, 2025-01-20:
Hello,
first of all I want to say the new "large" version of the Heishamon is outstanding, really a good job.

However, one minor issue:

If I use an IP-scanner (specific: Advanced IP Scanner from www.advanced-ip-scanner.com) - it crashes the Heishamon reproducable.
Shortly after the start of a scan (and it is only relevant to scan the IP of Heishamon) the web console output stops. And due to the fact that I have enabled the simulation of the optional board, the Heatpump goes into error.

A reset and a restart of the heatpump clears the error.

I think it is not that big deal, but it is not nice to have the heatpump in error state in winter. There seems to be no possibilty to resolve the issue from a remote place, without physical access to the heishamon. 

Many thanks and best regards,

Juergen

naseweiss99, 2025-01-20:
I tested this with version 3.8 & 3.9 of the firmware.

geduxas, 2025-01-22:
hm.. strange tool.. it not only pings ip, but tries somehow identify  hardware by scaning it's ports.. so it could be some problems with network stack of heishamon,.. also it could be related to external library.. so it will require further investigation what is causing it to hang.. here is port list which are scanned with app:

TCP:
[image]

UDP:

[image]

ZonPV, 2025-01-28:
Same problem and solution over here. 
Not related to powersupply. 
The connected CZ-TAW1 is lost too.
Uptimes for more than 30 days, as long as I do not use that app. 
Using firmware V 3.8

Thnx for all the good work!

naseweiss99, 2025-03-02:
I believe it should stay open, as no solution so far.

naseweiss99, 2025-04-02:
still no solution.

closed 2025-05-08 by github-actions[bot]