heishamon/HeishaMon #633: Heishamon- have to repeat a setting few times to change it.

kaczkalolo, 2025-01-21:
Hi. I bought Small version i connected it to my HA. It works and i get all the sensors etc. but whenever i want to change something like for example mode of the heat pump let say from heat to heat + dhw i have to do this couple times. Same with any other setting- setting temp also reverts and i have to do this few times until it finally chamges. On the gui page wifi strenght is about 86%, correct received data is at 99% and mqtt reconnects at about 1000 when working for few days. What ca i do about it?

geduxas, 2025-01-21:
If i remember correctly settings are sent after packet from pump received, by default data from pump is fetched every 10 or 15sec. So wait for it.

IgorYbema, 2025-01-21:
Yes it is normal that after a command you first get an answer with the old value. The second answer will contain the modified answer.

kaczkalolo, 2025-01-21:
Ok i tried, i clicked mode Heat only (it was set to heat + dwh) waited for a minute and than HA app refreshed on my phone and no go - Heat + DWH...

geduxas, 2025-01-21:
Where you had clicked? Also console output would make more clear.

IgorYbema, 2025-01-21:
To make sure this isn't the HA pluging causing the issue please follow the heishamon console log during the change. It should tell you more if the command was received by heishamon and if it was sent to the heatpump.

And are you sure you don't have the CZ-TAW1 overwriting the same setting?

MiG-41, 2025-01-22:
Or automation you created makes this ( since it expect imidietly confirmation ,that change is applied , but for first or second answer it recieves still old value , so correct it ...)
So observe/log  also commands topics in MQTT-Explorer ( or similar program with can monitor in realtime MQTT ).

Zwen1111, 2025-01-22:
I'm using the "small" version with firmware version 3.8 running. Last few days I noticed the same behaviour and found out that the connection is lost for a couple of seconds and reconnects. After a reboot the amount of reconnects has reduced, but changing the mode worked immediately via MQTT

Binifada1956, 2025-01-23:
TOP26 - Defrosting_State