heishamon/HeishaMon #634: Model Number not shown correctly

Skol6, 2025-01-21:
HI,

in the latest 3.9 SW the model number is shown in HEX instead of text.

Heat_Pump_Model = C2 D3 0B 34 65 B2 D3 0B 95 65

please fix it.
Thanks

geduxas, 2025-01-21:
From now it will look in hex, so it's endpoint response to map model by it's hex.

IgorYbema, 2025-01-21:
Did you read the release notes? This a feature and not bug :-)

Reason is that it is impossible to add all models into the code. It will only cause the code to get larger and use memory but it doesn't add any function. It is only a representation. It doesn't matter if the model is decoded or not. 

So we choose to only show the model bytes and let the end-user software to decode it if needed.

Skol6, 2025-01-21:
sorry my fault!!

closed 2025-01-21 by Skol6