heishamon/HeishaMon #637: Update to 3.9 still shows version 3.8

Tom54497, 2025-01-25:
Version number is not correct. What is a neopixel fix? Is that the complete version?

IgorYbema, 2025-01-25:
If it shows 3.8 then it didn't update yet. Try again and do a reboot first before updating. This sometimes help. 
The neopixel fix is only the last commit in this release. You can find the whole release notes in the releases section

Tom54497, 2025-01-25:
Rebooted before update. Process shows update successfull, but version is still showing 3.8

IgorYbema, 2025-01-25:
Are you uploading the correct file for your board type?

Tom54497, 2025-01-25:
Yes, large Version PCB and process is showing update successful but isn't

IgorYbema, 2025-01-25:
And which file are you using?

Tom54497, 2025-01-25:
HeishaMon.ino.esp32-v3.9.bin 1,36 MB

IgorYbema, 2025-01-25:
Ok so you are sure you upload the binary file and didn't download the html content of the github site for that file? I ask that because some people did make that mistake.
You need this download: https://github.com/IgorYbema/HeishaMon/raw/refs/heads/main/binaries/model-type-large/HeishaMon.ino.esp32-v3.9.bin

Tom54497, 2025-01-25:
Now tried to update with the file you sent and filled out MD5. Same. Update successful but still 3.8

Tom54497, 2025-01-25:
Is there anything in the 3.9 version how I can detect the real status like new TOP's? Might be that only the version number is not updated!?

IgorYbema, 2025-01-25:
TOP138 is the latest there

IgorYbema, 2025-01-25:
Can you try uploading this file (after unpackking from zip) using espressif.github.io/esptool-js/ ? Make sure you connect the heishamon using the usb-c to your computer and change address from 0x1000 to 0x0000
[HeishaMon.ino.esp32-v3.9.complete.zip](https://github.com/user-attachments/files/18545581/HeishaMon.ino.esp32-
v3.9.complete.zip)

Tom54497, 2025-01-25:
Current firmware 3.8 ends with TOP118

What is the MD5 for that file you provided? Is it a must to update via the tool because I do not have access to heishamon now.

It also has a different size of 1,46 MB

IgorYbema, 2025-01-25:
Yes this is a file for complete flashing instead of using heishamon interface.
I don't have calculated the md5 but just unpack the zip and then upload using the ESP tool.

I don't understand why the firmware update isn't working for you using the normal way. Can you try to downgrade to an older verison to see if that works?

Tom54497, 2025-01-25:
Downgrading to 3.7 works flawlessly as usual. And upgrading from 3.7 to 3.9 now works also. But from 3.8 to 3.9 not. That might help for error research! Thanks for your help.

Tom54497, 2025-01-25:
Did that without the tool with the files on github now.

IgorYbema, 2025-01-25:
Ok fine that it worked now. Don't know why this happened

closed 2025-01-25 by Tom54497