heishamon/HeishaMon #642: Add WH-MXC12H9E8 to HeatPumpType.md

smshomey, 2025-01-29:
Hi all,
I just updated Heishamon to v3.9 and would like to ask whether it's possible to add my heatpump model to HeatPumpType.md as well?

Here are my model details:

TOP92 Heat_Pump_Model: E2 CF 0B 44 34 12 D0 0C 34 35

IDU: Monoblock
ODU: WH-MXC12H9E8
KIT: Monoblock
Power: 12 kW
1ph/3ph: 3ph
T-CAP/HP: T-CAP

Thanks a lot!

IgorYbema, 2025-01-29:
You can do this yourself. Go to https://github.com/Egyras/HeishaMon/blob/master/HeatPumpType.md and hit the edit button. Then change the file and propse the change as a PR to this repo

smshomey, 2025-01-29:
Ok, thanks for the hint. I'm not familiar with the process but tried my best to issue a PR here: https://github.com/Egyras/HeishaMon/pull/643
I hope it fits.

IgorYbema, 2025-01-29:
Perfect. You even noticed that number 50 was missing and used that spot. Good one!

closed 2025-03-07 by github-actions[bot]